supermerill/SuperSlicer #384: Bad gcode preview with big files

Coffee0297, 2020-08-04:
### Version
2.2.52.1

### Operating system type + version
Win 10

### Behavior
When slicing a big stl the gcode preview is broken.
[image]
[image]
[image]

when viewing in ideamaker or any other viewer the gcode is fine.
[image]

#### Project File (.3MF) where problem occurs

[Engine_Block_-_Scaled.zip](https://github.com/supermerill/SuperSlicer/files/5024163/Engine_Block_-_Scaled.zip)

n8bot, 2020-08-04:
Also present in PrusaSlicer. It seems related to the secondary visualization, after the GCode is re-parsed. The more small segments there are, the worse the problem gets, but it only occurs after a certain number of segments/layers. At the beginning/lower layers of the print visualization, no gaps seem to appear.

supermerill, 2020-08-04:
mmm, they are rewriting this part in their alpha. I'll may wait after the merge before investigating that.
but thank you, I waited a long time for this kind of "big file issue" to come here. Now i have a .3mf to work with.

n8bot, 2020-08-04:
Nice. I was not aware they were already re-writing that part. I'm sure they'll fix it/are aware of it. If not, maybe make the issue in the main Repo?

Coffee0297, 2020-08-04:
If you need i can supply some more .3mf files where it happens? Atleast 1

Coffee0297, 2020-12-16:
[skull_broken.zip](https://github.com/supermerill/SuperSlicer/files/5704679/skull_broken.zip)
so its also broken in 2.3.55.2
[image]

Coffee0297, 2022-02-18:
> mmm, they are rewriting this part in their alpha. I'll may wait after the merge before investigating that. but thank you, I waited a long time for this kind of "big file issue" to come here. Now i have a .3mf to work with.

now 1.5 years later, do we have any fix in sight? ive not tested PS with same .3mf

supermerill, 2022-02-18:
Try to put a `G92 E0` in a layer change custom gcode.