supermerill/SuperSlicer #387: Gcode didn't reflect preview

yanfali, 2020-08-06:
### Version
2.2.52

### Operating system type + version
SuperSlicer
Version:   2.2.52
Build:     SuperSlicer_2.2.52_2020-07-03

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Mojave Version 10.14.6 (Build 18G6020)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 INTEL-12.10.22
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Plus Graphics 650
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

### 3D printer brand / version + firmware version (if known)
CR10S - heavily modified skr 1.4 turbo, blt, 0.9 degree motors on x,y,e, DD Bondtech, V6 HE

Klipper commit 1c508c38aa424f914c4c9843330e0f4849e4d26f (HEAD -> master, origin/master, origin/HEAD)

### Behavior
* _Describe the problem_

Slice the included STL file. Look at preview. Print at 0.8mm detail with a 0.4mm nozzle. It shifted only the support in the print. Nothing else. Very strange. Everything in print looks great.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

[image]

[image]

[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

File is 20MB, so I'm uploading a zip file to drive with STL, 3mf and gcode in it

https://drive.google.com/file/d/1O7qMughU7OBrBaWV0y4vjyPYaVf48hd0/view?usp=sharing

yanfali, 2020-08-06:
[image]

used the support in the file, not generated.

yanfali, 2020-08-09:
OK, just had something similar happen. Gcode looks fine but printed same file twice and got the same result. I don't think my motors are slipping. I upped voltage 10% and slowed accel 20% for second print https://drive.google.com/file/d/1LXm8tBBXzBYJ4qEJ4RuC9NlY3FIzER5I/view?usp=sharing

yanfali, 2020-08-09:
[image]

yanfali, 2020-08-09:
Attaching my settings in case it's something I did wrong
[settings.zip](https://github.com/supermerill/SuperSlicer/files/5046598/settings.zip)

supermerill, 2020-08-11:
[image]
no obvious problem on gcode

yanfali, 2020-08-11:
Yeah. Thanks. I think I have some motor issues or skipping.

yanfali, 2020-08-11:
Found a loose pulley, so I think the 2nd report is bogus. Testing it now. First report is on different axis and still valid I believe. Thanks for your help.

closed 2020-10-29 by supermerill