supermerill/SuperSlicer #388: Z translation is not allowed

zbrozek, 2020-08-07:
### Version
2.2.52

### Behavior
SuperSlicer has a field to translate a part in Z, but if you type in a value it is immediately reset to zero and the part is clamped to the bed. I would like to be able to use this field to print a part above the bed.

I'm not sure if this is a bug or a feature request.

liftbag, 2020-08-07:
It's not a bug, but a normal behavior, inherited from PrusaSlicer. It could be a preventative measure to avoid throwing a print for inadvertently detaching the piece from the bed in the slicer.
Of course you may want to lay a second piece on top of the main one, and you can do by adding it as part of the first. Superslicer also has a useful function to allow you to print stacked objects and separated from each other, avoiding the error that PrusaSlicer shows when two pieces are completely separated in the Z axis.

supermerill, 2020-08-07:
basically like liftbag said.
You can use this box on parts that are in a multi-part object.

supermerill, 2020-08-24:
A little update
I tried to allow random Z, takes me some hours to acheive and the result is buggy because many thing assume that the object IS on the build plate.

So currently, it's shelved.

supermerill, 2020-08-25:
If someone want to continue the work, the changes are in branch https://github.com/supermerill/SuperSlicer/tree/arbitrary_z