supermerill/SuperSlicer #393: Feature request: Gyroid infill, "don't connect only to the external perimeters"

robustini, 2020-08-09:
For the parts I'm printing it would be very convenient if the Gyroid infill had an option to not connect it only in the outer perimeter of the piece, but in the internal structures yes.
Now it is possible to enable "don't connect" but it is obviously also disconnected in any internal structures.
I'm printing wings and this option would save me a lot of weight while maintaining solidity in the internal structures.

SupserSlicer V2.2.52
Windows 10 x64
Delta printer custom build

robustini, 2020-08-09:
"Don't connect" enabled in the first image, and disabled in the second.
You can see that the seat for a tube is wound thick, almost doubled in thickness, and that is fine, but I'd like the outer edge to remain as with "don't connect" enabled instead.

[image]

[image]

supermerill, 2020-08-17:
I can create a box that let you choose from:
- connect to everything (default)
- don't connect
- connect to outer shell only
- connect to holes (inner shell) only

that's that?

robustini, 2020-08-31:
@supermerill when do you think you can make these changes?

supermerill, 2020-08-31:
I don't need it personally, so I don't know.
If you need it quickly, you can pay me for it.

closed 2020-10-22 by supermerill

robustini, 2020-12-11:
Thanks @supermerill!