supermerill/SuperSlicer #395: Z-Lift on Top/First layers ONLY

Alonso3D, 2020-08-10:
### Version
Future versions

_Is this a new feature request?_
Yes!

Check box(es) to activate Z-Hop on first and top layers only. For top layers it would avoid leaving scars and save any extra wear and tear on the printer through out the rest of the print job. 

Perhaps also include a parameter for how many layers to activate Z-hop at the start end of the objects.

For example, activate Z-hop only the first 2 layers of the print job and activate Z-hop on the LAST 2 layers of the object.

robthide37, 2020-08-12:
isn't this already available with the only lift z above and only lift z below paramter

supermerill, 2020-08-12:
"top" is an extrusion type, it can happen at every layer.

Kirinkunhime, 2020-10-27:
I +1 this feature request - having Z-Hop only on some layers would allow for some interesting options like (limited) multi-color prints without the need of filament change or an MMU.

Matts-Hub, 2020-10-28:
My apologies for the duplicate, +1 for this feature request!

closed 2020-11-25 by supermerill