supermerill/SuperSlicer #396: Insert Custom Gcode: insertion errors over multiple edits

rbtwas, 2020-08-10:
SuperSlicer 2.2.52
Linux 18.04

At first, when inserting custom gcode, things worked ok. This was done over multiple project saves.

But now, after inserting gcode (3 separate points) and after doing *Slice Now*, SuperSlicer *moves* the insert points to different
layers. 

Also, sometimes SuperSlicer, even though the code does not appear to be moved in the gui, the code is clearly being inserted
*after* the target layer, not before.

The three inserts should have been at: 1.45, 3.05, and 3.45, but are shown at 1.53, 3.25, and 3.45. 

Looking at the actual gcode the inserts are at: 1.85, 3.65, 3.810.

Tried this with prusa. Not seeing the same behavior.

(".txt" added to get past the fileNaughtSee.

[3dpp.20200809.1919.mann.h2fj.bug.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5048819/3dpp.20200809.1919.mann.h2fj.bug.3mf.txt)
[3dpp.20200809.1919.mann.h2fj.bug.gcode.txt](https://github.com/supermerill/SuperSlicer/files/5048820/3dpp.20200809.1919.mann.h2fj.bug.gcode.txt)

supermerill, 2020-08-11:
Maybe it's the same problem with #372, i'll test with the latest daily build.

rbtwas, 2020-08-11:
It looks like it's not forgetting the old inserts, and gets confused on the
offset computation.

supermerill, 2020-08-11:
1)
0.35mm on the first layer
then 0.2 mm for each layer
so you have layers at 1.35, 1.55, 1.75 ... you can't have a change at 1.45!

2) you set a z-offset of 0.3, so the 1.55 translate into a 1.85

It's difficult to understand and set, because you have rafts and offset

rbtwas, 2020-08-11:
I'm not generating these numbers, SuperSlicer is. The numbers come
from the layer chooser.

Also, when changing layers, sometimes the old layers don't go away. I
have 3 layers changes, but I end up with 4,5, and sometimes six.

It did not do this initially. This is some  kind of cumulative error.
It appears that layers in the raft change, then the insert numbers are
somehow
applied after. The code is actually going into the wrong place! I
noticed this in the  print itself. I was trying to figure out why my
inserts were
appearing at the wrong layer *in the print*.

Another error: sometimes the code is inserted after the chosen layer
gcode instead of before.

So you insert code at layer 7 (enumeration), the code is going in
after layer 7 code instead of just before like it is supposed to.

To be clear, Superslicer says it is inserting at layer 7, which means
user gcode is supposed to go in before layer 7 gets executed. But
sometimes
it goes in after layer 7 code!

I tried to figure out why it was doing that but never could. I had to
start all over (clear the project file, and generate a new one) to get
it clear.

These errors only started appearing after about the 5th re-edit or so.
I may have to try to characterize the problem by trying to reproduce
it.

I eventually gave up and had to put inserts in manually with an editor. :/

supermerill, 2020-08-11:
I don't understand.

The problem is that the raft don't move the pre-applied gcode inserts?

rbtwas, 2020-08-11:
Ok, I created a little doc to help explain what I am seeing. I
apologize in advance, I'm no master Libre user.

Please find attached pdf.

supermerill, 2020-08-12:
i don't see any attached pdf

rbtwas, 2020-08-12:
I"m guessing github has blocked it. Lemme try posting it directly.

rbtwas, 2020-08-12:
Pdf
[sprSlcr_bugRpt.20200811.1554.pdf](https://github.com/supermerill/SuperSlicer/files/5061308/sprSlcr_bugRpt.20200811.1554.pdf)

closed 2021-06-14 by supermerill