supermerill/SuperSlicer #397: External perimeter first on first layer

guestisp, 2020-08-10:
Most of the time, printing the first layer from outer to inner perimeters has only advantages (a longer path will stick better on the bed than a smaller one and so on).

Actually , for every print, i add a modifier to change the print order on the first layer.
A dedicated option would be ok. Even better would be to set a numer, telling at which layer the "normal" order would be used, in example:

`External perimeter first for` `4` `layers`

will print from external to internal the first 4 layers (probably the only useful option value would be 1).

KimmoHop, 2020-08-10:
Yes, this would be nice!
IdeaMaker has per-layer settings, which is already easier than using modifier for each object, but one counter/checkbox would be better ;)

supermerill, 2020-08-11:
> per-layer settings

like the layer range modifiers?

[image]

guestisp, 2020-08-11:
I think we are talking about two different things.
Mine request was to add a simple checkbox or a simple input field , saying to print external perimeter first on the first layer only or up to layer X. Much easier and faster than having to modify, each time, the object before print.

A print setting will be stored in the profile, so every time I have to print, the first layer can be printed in reverse order. (that is what I have to do every time on every print)

supermerill, 2020-08-11:
> saying to print "option X" on the first layer only or up to layer X

There is already so much settings "only on the first layer".
Maybe i should add a page to create "for the first layer" / "for the X first layers" modifiers that will be applied on the print.

JoduWei, 2020-08-17:
@supermerill I have big plates with many parts. Is there a way to set a height range modifier for all parts?

supermerill, 2020-08-17:
Nope

KimmoHop, 2020-08-18:
The setting could activate 1st layer perimeter order reversing, though I haven't yet checked what it really produces in gcode. Currently reversing is done when brim is used (PerimeterGenerator.cpp, after line 730)

I'd like to look at this, but I can't fork SS. I have forked prusa3d/PrusaSlicer and when I (try to) fork SuperSlicer, github shows my PrusaSlicer fork :/

n8bot, 2020-08-18:
I think if you make a new branch on your fork you can rebase it to one of SuperSlicer's branches and you will then have a working branch of SuperSlicer.

supermerill, 2020-08-18:
prusa slier and superslicer are the same repository, just not the same branches.

VanessaE, 2020-08-31:
This would be a very useful feature indeed, but make it more generic:

On the first layer (or first N layers), sort all perimeter loops by length, printing them in descending order by length wherever possible, regardless of whether they're external or internal perimeter loops.  After all, on the first layer we don't care about X/Y surface quality (as there's effectively no surface yet to worry about :stuck_out_tongue: ), but instead we're concerned with the Z direction, i.e. getting everything down onto the build plate as cleanly as possible.

I suggest this because the external perimeters won't always be the longest.  For example, if the print has a vertical hole or some kind of downward facing embossing (as with the lettering on the bottom of a Benchy), divots, etc., then those features' *innermost* perimeters (as in, the loops adjacent to the infill) will be largest, not their external ones.

supermerill, 2020-08-31:
I think it's better to print a perimeter next to an already printed one than a big perimeter (temporary) isolated.

VanessaE, 2020-08-31:
That may be better to some extent, but you have to start somewhere; if you're starting a new set of perimeters, start with the longest one, so that you're not drawing a bunch of tiny islands that'll tend to want to peel up. :smile:

ProfEngr, 2020-10-05:
For a specific example I have a thingiverse ear saver with lettering cut through the middle of the band. The small details for the lettering would adhere much better if the path planner would put down the longer (next to the infill) traces first before trying to squirt out tiny blobs that may or may not stick to the build surface. If they don't then they usually get dragged around and deposited somewhere else that's less than desirable.

supermerill, 2020-10-05:
@ProfEngr you want the external peimeter be printed last, right? remove the brim and it should reverse.

ProfEngr, 2020-10-06:
I figured out how to manage this in the specific STL I mentioned. I set external perimeters first as a part modifier then added a height modifier at layer 2 and left the same option unchecked. The gcode preview examined in another slicer (PS/SS really need a line-by-line preview to go w/ layer-by-layer) indicates I'll have the desired print order, starting at layer 3. Not sure why the height modifier set for 0.4-2.0mm would reverse the order at 0.6.

guestisp, 2020-10-20:
> That may be better to some extent, but you have to start somewhere; if you're starting a new set of perimeters, start with the longest one, so that you're not drawing a bunch of tiny islands that'll tend to want to peel up

Exactly

ProfEngr, 2020-10-21:
So, I'm reprinting a cooling duct and tried the height range modifier again. It failed to perform as expected. I'm either getting double gcode on layer 1 such that the nozzle retraces the whole layer while extruding or if it gets that right it ignores my setting for 1mm bottom thickness (only prints 1 base layer) even if I set 5 bottom layers as an override.

supermerill, 2020-10-21:
gimme your project file

ProfEngr, 2020-10-21:
@supermerill does the API data get exported to the file?

supermerill, 2020-10-21:
yes, this will be corrected in the 2.3.55, as prusa has fixed that by moving it to a different setting space.
So for the time being, it's safer to remove your IP or key before saving the file for sharing. But if it's a local network IP, no worry, it's in practice inaccessible from the outside.

ProfEngr, 2020-10-21:
Removed the data then resaved the 3mfs. 
Thin base was he first slice for testing. It generated 1 base layer before starting infill on layer 2. The path went as expected otherwise w/ outside first on the first layer then inside out for the rest of the part.

The other 2 are labeled for the error (as I see it) in gcode output.
Double Passes should be self-explanatory. It hits each layer twice w/ layer 2 having different perimeter counts.
Skip layer has the correct perimeters for 1 and 3, but the 2nd has the default from the tab ignoring the modifiers I applied.
[layer height modifier.zip](https://github.com/supermerill/SuperSlicer/files/5418755/layer.height.modifier.zip)

Also, I noticed the tooltip for the Z Full Step setting in the Printer tab says enter 0 to disable. This generates an out of bounds error message.

burner-, 2021-02-07:
This would be awesome. Also some kind "round features first" would be good. I have currently promlem with object what have sharp turns and big loops that it print sharp turns first and then knock them off with circular perimeter pattern. 
[image]

guestisp, 2022-10-19:
@supermerill any news on this  ? printing internal to external on first layer is almost useless in any case and lead to issues most of the time (little extrusions on very short perimeters). almost all these issues can be solved at once printing from external to internal (thus, from the bigger to the smaller perimeter) on first layer.

guestisp, 2022-10-19:
> This would be awesome. Also some kind "round features first" would be good. I have currently promlem with object what have sharp turns and big loops that it print sharp turns first and then knock them off with circular perimeter pattern. [image]

I think that on first layer, the print should be done in any way that guarantee printing bigger perimeters and bigger features before the smaller ones. Everything is good as long as you are extruding as much as possible sticking to the bed. Then small features are easier to print because they have an already printed perimeter to stick to in addition to the bed.