supermerill/SuperSlicer #399: Outline Perimeter Brim Ears

tcurdt, 2020-08-11:
### Version

2.2.52

### Operating system type + version

macOS 10.15.3

### 3D printer brand / version + firmware version (if known)

Prusa MK3 Bear

### Behavior

The Brim Ears are fantastic. It's what I sometimes do manually, too.
Unfortunately the way they are currently sliced often causes problems.

[image]

[image]

The many changes in direction without an outline perimeter too often causes the ears to get loose.

I would rather have an option for them become part of the object like this

[image]

(although I am not sure why there is such a dent of the inner perimeter)

or at least have a dedicated outline perimeter.

[image]

#### Project File (.3MF) where problem occurs

[example.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5056717/example.3mf.zip)

supermerill, 2020-08-11:
I can change the infill from concentric to concentric_with_perimeter

but i never had a loose ear. It's not an adhesion problem on your printer?

tcurdt, 2020-08-11:
> I can change the infill from concentric to concentric_with_perimeter but i never had a loose ear.

That or making it configurable would be great.

> It's not an adhesion problem on your printer?

Well, I usually don't have bigger problems with adhesion.

All I can say is that printing failures have spiked for me when using these kind of ears.
The manual ears never cause any problems.

The filament brand and summer temperatures could have an effect?
Maybe the slightly different extrusion on the first layer could also contribute?
I don't really now.

But it seems a perimeter would just help.

> I won't "integrate it" in the part because it makes it impossible to remove by hand.

I understand the idea, but "by hand" never worked well enough for me with either.
In reality I still use the same tool to remove them for both.
So at least for me that's not a requirement. Just saying.

yschroeder, 2020-08-16:
Isn't the purpose of adding brim ears to deal with adhesion problems? The best of both worlds would be to make the brim ear being concentric perimeters. Would be as easy to snap off as the current solution. I drew something by finger on my phone to illustrate. 
[image]

supermerill, 2020-08-16:
Concentric create gaps, so the perimeter + rectilinear infill should be more useful (the same as the "dense infill" pattern)

closed 2020-10-22 by supermerill