supermerill/SuperSlicer #401: Disable fan for top fill

matthewfallshaw, 2020-08-11:
### Version
2.2.52

### Behavior
* _The Problem_
    The Filament Settings > Fan speed > Top fan speed *setting* is *disabled* by setting it to 0, but I want to *enable* the setting to *turn off the fan* for top layers… so I want to be able to set the setting to 0 and have the setting *not* be disabled, so that the fan is turned off for top layers.
* _Steps needed to reproduce the problem_
  * Form an intention in your mind to disable the _top fan speed_ setting. Act on that intention.
  * Form an intention in your mind to enable the _top fan speed_ setting to turn off the fan for top layers. Attempt to act on that intention.
  * Notice that there was no change you could make to your SuperSlicer settings to act on the second intention. 😄 
* _Expected Results_
  * On forming the second intention there is some action you can take to change SuperSlicer fan speed settings.
* _Lame Suggestion_
  * I really don't love this idea, since we have a strong convention that `0` disables settings, but if this setting (or all settings?) were disabled by `-1` instead of `0`, then this problem would be resolved. I note that this would also resolve some other lameness such as having to use `180` in Print Settings > Infill > Advanced > Angle > Bridging to achieve a 0° angle.

supermerill, 2020-08-12:
in the meantime, you can set it to 1%, it should be low enough that the fan won't move.

matthewfallshaw, 2020-08-15:
How hard is it to have the installer / upgrade script notice if VERSION < X and config value == 0, and set config value to -1?
*If that is easy to do*, then it seems a neat way not to feel constrained by past config — change the old value that means DISABLE to the new value that means DISABLE, and behaviour won't change for any user.
If it's hard to do, then I do not presume to imply that you should do it. Thank you for all that you have already done.

supermerill, 2020-08-16:
it's possible to create a new option and deprecated the old one, and create a bit of code for conversion.... I'll look into it.

closed 2020-11-25 by supermerill