supermerill/SuperSlicer #402: Ironing patterns

ciro1138, 2020-08-12:
I have a few ironing pattern enhancements.
 
1. It would be nice to have an option for a spiralized ironing pattern to use with round parts.

2. This is a thought from CNC machining for clearing pockets. What it does is a looping path that follows a "wall" and doubles back over the same area. This could allow for a smoother surface faster ironing and self clean any over extrusion spots by getting pushed from the inside of the looping pattern with the nozzle as it does the following loop.  Reference: [https://cadcamlessons.com/wp-content/uploads/2019/03/2d-adaptive-clearing-fusion-360-11.png](url)

supermerill, 2020-08-12:
1. if you want ironing to use concentric over rectilinear,  it's not a good idea because concentric create gaps.

2. nice idea, maybe someone can modify rectilinear to create a new pattern like that.

ciro1138, 2020-08-12:
> 2. nice idea, maybe someone can modify rectilinear to create a new pattern like that.

Any starting pointers? I don't mind giving it a try, not great with programming but I am willing.

supermerill, 2020-08-12:
1) cloning the repository and succeed to compile & run it.
2) create a new infill, by copy-pasting an existing one, like Gyroid or 3D honeycomb or FillScatteredRectilinear. To do that search every file where the label appear and create a new one. There is the class, the enum definitions, and the array of label to enum.
3) test it, then modify the fill_surface / fill_surface_exrtrusion method to change the pattern by creating your own extrusion.
4) understand the rectilinear2 code (or the rectilinear one, which is an older simplier verison) and create a different version to implement the wanted pattern.

n8bot, 2020-08-12:
Those are great suggestions.

What do you mean that concentric infill creates gaps? Because the spacing is adjusted?

supermerill, 2020-08-12:
compare 'concentric' and 'concentric with gapfill' patterns on a difficult surface, you'll see.

ciro1138, 2020-08-13:
Preferably I would not do a concentric pattern,a spiral pattern would be the method of choice. Then you have a small gap in the begining and end. Those could be taken care of with a over extruding gradient gap fill in the begining and end.

n8bot, 2020-08-13:
> compare 'concentric' and 'concentric with gapfill' patterns on a difficult surface, you'll see.

Ah, yes, I see what you mean:
[image]