supermerill/SuperSlicer #406: Perimeter Dent

tcurdt, 2020-08-14:
### Version

2.2.52

### Operating system type + version

macOS 10.15.6

### 3D printer brand / version + firmware version (if known)

Prusa MK3 Bear

### Behavior

In order to have brim ears (see #399) I am using cylinders. In Prusa Slicer they are getting sliced as expected as such

[image]

In SuperSlicer I am getting a weird dent.

[image]

Not sure if or what setting this could be that I missed. Otherwise I would consider this a bug.

#### Project File (.3MF) where problem occurs

[perimeter.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5074417/perimeter.3mf.zip)

tcurdt, 2020-08-14:
By turning off "only one perimeter on top surfaces" I am getting the prusa slicer behaviour at least.

neophyl, 2020-08-14:
That actually makes sense to me (after the fact of course).  The 'top infill' is inset into the rest of the model to bind the layers together which is what its doing on the 1 perimeter version where the other is just filling in all with perimeters.  

The problem is that as your mouse ears are part of the model it has no way of knowing that they are 'disposable'  and not part of the actual design.

supermerill, 2020-08-16:
In expert mode, on the object -> height range modifier -> add settings -> perimeters & shells -> only one perimeter on top layer -> uncheck it.

tcurdt, 2020-08-17:
Although now I understand why it happens - it's still weird.
But since there is a work around - let's close it?

closed 2020-08-17 by supermerill