supermerill/SuperSlicer #407: Thick bridge

espr14, 2020-08-14:
### Version
v2.2.51

### Operating system type + version
Win 10 64

### 3D printer brand / version + firmware version (if known)
all

### Behavior
I have nozzle 0.4 mm. I set width of (external) perimeters to 1 mm. But all the bridges are extruded like I have 1 mm nozzle so they fall immediately. Can you reduce the flow to nozzle diameter for bridges?

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs

yschroeder, 2020-08-16:
Can't check right now, because I am on mobile so this might be wrong:
Bridges have their own extrusion width setting. If you enable "detect bridging perimeters" this setting is also used for perimeters.

supermerill, 2020-08-16:
@espr14 
i just checked the code, and bridge should use the nozzle diameter. Can you link your .3mf project file? (save as, zip it and drop it here)

n8bot, 2020-08-16:
The bridges are also extruded like a cylinder, not a squished extrusion. Their "height" is the diameter of the nozzle width.

espr14, 2020-08-20:
I tried bridge test and it looks good. Looks like it was due to short bridge and not tuned K factor. High pressure caused overextrusion.

espr14, 2020-08-20:
Still, there should be only one perimeter on the bridge. The bridge infill should count with different width of thread and diameter of the nozzle. If wide thread is used, the infill does not touch the bridge perimeters.

supermerill, 2020-08-20:
I don't understand.

Also, i think there is a problem with bridge, it's that there is no pressure relief at the start of the bridge line. But for that, i have to create a pressure model.

espr14, 2020-08-20:
[image]

n8bot, 2020-08-20:
It looks like you have your nozzle diameter set quite small, with the extrusion widths quite wide. The bridges always print as a cylinder with the diameter of the nozzle width.

supermerill, 2020-08-20:
to avoid that, you can use "no perimeter on bridge area" -> "keep only bridges"
I can do an other option if needed.

espr14, 2020-08-20:
Finally, I found it but it does not work in this case:
[no perimeters bridges.zip](https://github.com/supermerill/SuperSlicer/files/5104425/no.perimeters.bridges.zip)

supermerill, 2020-09-06:
I loaded your model but i don't see an anything wrong.
Can you post the project file and show me a picture of where the problem is?
I can' spot a bridge or overhang that aren't extruded with nozzle diameter.

espr14, 2020-09-07:
Default setting. Looks good but there's a gap between perimeters:
[image]

Keep only bridges. No perimeters but part is missing.
[image]

supermerill, 2020-09-07:
> there's a gap between perimeters

I don't know how to resolve that. It's not possible to "add perimeters", and if you use bridge flow for overhangs, it's defined by your bridge flow setting...

yschroeder, 2020-09-15:
Maybe it would be possible to move the perimeters to the outside edge instead of adding more and fill the space that is then opened up with regular bridge infill?

supermerill, 2020-09-15:
This would be very expensive (in difficulty and computation)