supermerill/SuperSlicer #409: Feature request: Project workflow

imbaimbaimba, 2020-08-15:
More often than not, when 3D-printing for bigger projects, you need to prepare multiple prints. I think it would be great to have a slicer that made this whole workflow a lot easier.

I imagine something like a project tab that deals with all this. Here I imagine having a "New Project" button, that I could press and name a project, maybe give it an icon etc. Inside this project, I imagine being able to drop all my STL files for the project, and here set specific overrides for them like infill percentage etc. From there, I would like to be able to assign them to build plates, that could then appear in tabs for each print to complete the entire project. Each build plate in the project could then have names and settings of their own, so one build plate could for example be for PETG prints, while another could be for TPU prints. There are a lot of other additions/features that could be implemented into this, like automatic distribution to build plates how they best fit according to settings etc. and more, but I am mainly trying to outline the general idea.

A system like this would give me an amazing overview of the projects I am building. Also, being able to save everything down into a project file like this, to share with others, would mean others could get going on a project extremely fast.

Is this completely unrealistic?

supermerill, 2020-08-16:
Currently the project file is for one platter only.
things to change that comes to my mind:
- button to add multiple platter tab
- can change tab name
- tie all ui elements to the current platter, and not the "global" one that is now deleted (that is the big thing).
- save & load the platters & objects platter assignment in the .3mf
- a way to move an object to an other platter (right clic?)

what do you think?

imbaimbaimba, 2020-08-16:
Seems like you have catched my idea very well!

Of course, you gotta excuse me for having no idea how easy/hard stuff like this is to implement, and what is actually possible, so I am just suggesting straight from imagination. Your changes sound completely in line with what I am suggesting, and from there it could of course be built and improved upon.

I can maybe try and throw together a quick sketch of how I imagine the layout could be? If you don't mind an ugly paint drawing ;)

n8bot, 2020-08-16:
I have a suggestion related to workflow:

Perhaps a saved profile could include height range modifiers or other settings-based (not geometry-based) modifiers.

supermerill, 2020-08-16:
"profile" is not a  "workflow"
"workflow" here is more like a "project".
a project store the objects and modifiers.
a profile can't store modifiers because they are tied to specific geometry.

Maybe you ask for "global" modifiers? (that doesn't exist and should require a different issue), that can be stored somewhere and applied on a plater on-demand?

n8bot, 2020-08-16:
I see what you're saying. Definitely an unrelated issue. Sorry. I'm very bad at determining relevance sometimes.

I think your understanding is correct about what I meant. The only use-case I had in mind was having the second layer height also be able to be saved into the profile.

I think there is a lot of value in a more robust project-based workflow, and I'm excited to see imbaimbaimba's sketches. Sorry for derailing another topic.