supermerill/SuperSlicer #411: Retraction Calibration

prahjister, 2020-08-17:
Jeff Jiggens suggested I share a calibration process I have been working on and might want to integrate into your slicer.

https://github.com/prahjister/Calibration-Generator

I am preferring to go web based for cross platform compatibility but the bulk of code is available on github. I have only changed some defaults and formatting. I will push these Minor changes to github in a couple days.

You can check out web version here

prahjister.duckdns.org

I'm self hosting unless I can get enough donations to pay for it.

supermerill, 2020-08-17:
Thank you.
Your concept is good, but I can't integrate it as-is, as I don't generate gcode but a geometry.
My dev version currently doesn't care for speed, as i think it should be put to the maximum speed the extruder can safely handle, don't you think?

belm0, 2020-08-26:
@supermerill is a preview of your retraction calibration feature available?

supermerill, 2020-08-26:
[image]

belm0, 2020-08-26:
Using ridges to mark the changes is interesting.

Is this focused only on retraction distance?  I'm wanting to tune retraction and un-retraction speed, extra length on restart, wipe while retracting on/off (perhaps done as a separate pair of columns), and retract amount before wipe.  (Of course, only one parameter can be tuned at a time.)

For layer cooling time, I would leave it to the slicers default cooling handling, which I understand to slow down printing if the layer time is too fast.

belm0, 2020-08-26:
This article emphasizes the high complexity of retractions, and the importance of retraction / detraction speed.

https://www.sublimelayers.com/2016/10/some-musings-on-retracts.html

closed 2020-10-22 by supermerill