supermerill/SuperSlicer #413: Dark Theme

fernandex00, 2020-08-17:
### Version
2.2.52.0 Latest

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
twotrees bluer

### Behavior
* Feature Request: it will be nice for the eyes to be able to choose a dark theme, like other slicers

Is this a new feature request?: Yes

supermerill, 2020-08-17:
slic3r use wxwidget, and they only have dark theme for linux & macos.
Problem is the microsoft api that doesn't support it (yet).
It will comes when wxwidget supports it. And that will comes when microsft supports it.

Vlkvo, 2021-07-09:
As an visual impairment user i use Windows High contrast themes. Is it possible to add support for it? 

It is coming to PS soon, see: https://github.com/prusa3d/PrusaSlicer/issues/3038

btw, there is also a new conversation about PS dark theme, looks it coming soon too: https://github.com/prusa3d/PrusaSlicer/issues/4493

supermerill, 2021-07-09:
if it's coming into prusaslicer, it will be merged when they will release it.

TWIISTED-STUDIOS, 2021-10-30:
Was this ever added I'm melting my eyes changing from Prusa to Super Slicer and I can't find anywhere to change it?

supermerill, 2021-10-30:
not merged yet. I wait for the first rc of the 2.4. I won't merge a beta build anymore, as it comes with bugs an I spend my time fixing bugs that will be fixed by prusa anyway.

TWIISTED-STUDIOS, 2021-10-30:
no worries than thank you for the fast reply, your slicer is very very detailed its taking some getting used to but the eye burn seems to be worth it

DanielDC88, 2022-05-21:
I think the latest stable version of Prusaslicer is now 2.4.2 - will this feaure be added now?

stevekenney318, 2022-06-03:
Still no dark mode ? I think this is a showstopper for me then.