supermerill/SuperSlicer #414: Feature Request: Overhang cooling

SnowDrifterr, 2020-08-18:
I think it would be useful to have variable cooling based on overhang angles. Some materials (PC, ABS, for example) suffer greatly in strength with even a slight bit of cooling. However, overhangs also have a tendency to curl (whereas straight walls do not). I'd like to see a feature that allows for cooling depending on overhang angle to allow for fanless printing through areas that don't need it.

supermerill, 2020-08-18:
you mean that youwant to remove the fan over bridges (because it's strait) and keep on overhangs (unless it's strait)?

I'm not sure that the cooling will reduce curling. Have you done a test with manually modified gcode?

SnowDrifterr, 2020-08-19:
Not necessarily touch bridges, but be able to set some sort of fan speed to overhang angle for cooling.

Have tested it and found that overhang cooling helps with curling. I'm getting around it right now with external perimeter fan speed but it's far from a perfect solution since PC in particular loses layer adhesion with any fan speed. Only cool where needed for tricky spots.

Noted the same behavior with petg abs as well. Pla to some extent but it doesn't suffer adhesion nearly as bad with reduced cooling

nyxaria, 2020-08-19:
This is implemented in Cura. What happens is that the fan is off all print, and only gets turned on when it is doing a bridge.

supermerill, 2020-08-19:
@nyxaria you can also do that here by using "bridges fan speed".

I think @SnowDrifterr want an other setting to work on overhangs only. And also, i don't know if the bridge fan trigger on overhangs.

SnowDrifterr, 2020-08-20:
Can confirm that bridge fan does not trigger on overhangs

Pavulon87, 2020-08-20:
It is possible to use "As bridge" overhangs, but sometimes even if set to 1% it is still normal perimeter.
[image]
It would be great if it worked like this:
[image]
Here is calicat rotated 2 degrees. 45 degrees is not considered an overhang.

supermerill, 2020-08-20:
@Pavulon87  fixed for next version

supermerill, 2020-11-12:
> Or just give it a extrusion role or something to make it happen on custom Gcode
It has it's own role

Stephan3, 2020-11-17:
@supermerill I checked today the role by inserting a:
`; [extrusion_role]`

The nightly build produces:

```
; custom gcode: feature_gcode
; ExternalPerimeter									<<-- extrusion role
SET_VELOCITY_LIMIT ACCEL=2000 ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
; custom gcode end: feature_gcode

;TYPE:External perimeter
G1 F213
M106 S51
G1 X165.980 Y167.307 E0.42863
M106 S201.45
G1 X169.277 Y167.308 E0.10862
M106 S51
M106 S201.45
G1 X169.528 Y167.340 E0.00834
M106 S201.45
; custom gcode: feature_gcode
; OverhangPerimeter								<<-- extrusion role
SET_VELOCITY_LIMIT ACCEL=2000  ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
; custom gcode end: feature_gcode

;TYPE:Overhang perimeter 							<<-- NOT triggering extrusion_role change??
G1 F2400.000
G1 X169.750 Y167.383 E0.00744
G1 X169.896 Y167.474 E0.00569
G1 X169.763 Y167.592 E0.00587
G1 X169.515 Y167.761 E0.00987
G1 X169.233 Y167.905 E0.01043
G1 X168.948 Y168.006 E0.00996
G1 X168.788 Y168.040 E0.00541
G1 X168.374 Y168.085 E0.01374
G1 X153.008 Y168.085 E0.50624
G1 X152.711 Y167.563 E0.01979
; custom gcode: feature_gcode
; ExternalPerimeter								<<-- extrusion role
SET_VELOCITY_LIMIT ACCEL=2000 ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
; custom gcode end: feature_gcode

;TYPE:External perimeter
G1 F213
G1 X152.928 Y167.347 E0.01009
SET_VELOCITY_LIMIT ACCEL=6000 ACCEL_TO_DECEL=6000 SQUARE_CORNER_VELOCITY=5		<<-- set by pp script to Travelacc
G1 X153.253 Y167.587 F30000.000
G1 X153.003 Y167.691
; custom gcode: feature_gcode
; GapFill										<<-- extrusion role
SET_VELOCITY_LIMIT ACCEL=2000  ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
; custom gcode end: feature_gcode
```

Its the github checkout of today. No idea if i did something wrong. Thats the output it produces.

supermerill, 2020-11-18:
```
; custom gcode: feature_gcode
**; OverhangPerimeter**								<<-- extrusion role
SET_VELOCITY_LIMIT ACCEL=2000  ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
; custom gcode end: feature_gcode

;TYPE:Overhang perimeter 							<<-- NOT triggering extrusion_role change??
```

I don't understand, the extrusion_role change isn't just a line above?

Stephan3, 2020-11-23:
> ```
> ; custom gcode: feature_gcode
> ; OverhangPerimeter								<<-- extrusion role
> SET_VELOCITY_LIMIT ACCEL=2000  ACCEL_TO_DECEL=2000 SQUARE_CORNER_VELOCITY=5
> ; custom gcode end: feature_gcode [...]

You were right, i just missed it. Sorry.

halfshavedyaks, 2021-09-10:
I don't often have problems with curlup, but I would still like to have some settings specially for overhangs - treated separately from normal perimeters and from bridges. 

To get a strong part with good layer adhesion I leave the fan off most of the time. For bridges I want a faster print speed , a lower extrusion rate, and 100% fan.

In contrast, for overhangs I want a fan that depends on the angle of the overhang (or at least a defined angle that starts the fan at a defined speed), a *slower* print speed and a normal extrusion rate/line width.

This helps with printing tidy overhangs without drooping, while still allowing maximum layer adhesion for the rest of the print.

The only slicer I know of that currently has this feature is the smartavionics fork of cura. However I'm not a fan of curs interface or clunky performance so I'm supporting this feature to devs of other slicers wherever I come across the topic.

Most slicers on github have a thread on this topic, but only smartavionics cura (afaik) has actually implemented a solution so far.

supermerill, 2021-09-27:
> a slower print speed

you can set overhang speed now

> a normal extrusion rate/line width.

set 0 in overhang bridge flow

For the fan speed, it's a bit complicated what you ask. Fans aren't responsive enough anyway.

levleon13, 2022-01-27:
> It is possible to use "As bridge" overhangs, but sometimes even if set to 1% it is still normal perimeter. [image] It would be great if it worked like this: [image] Here is calicat rotated 2 degrees. 45 degrees is not considered an overhang.

god i've been through ALL SLICERS POSSIBLE to realize there's no way of configuring overhang angle detection.
Is this something that might be added to super anytime soon? this will literally make my petg prints perfect.

supermerill, 2022-01-27:
> overhang angle detection.

search for overhangs_width_speed. It's not directly the angle, but the overhang width. To get the angle, you have to `arctan(overhangs_width_speed, layer_height)`

Garryb123, 2022-09-19:
"Why not have a warmup value? So how many ms the fan is to "warmup" and add that in advance so the fan is up to speed before it reaches the overhang?"

"Is this something that might be added to super anytime soon? this will literally make my petg prints perfect."

I too want overhang detection for the same reason as the OP. It will be perfect. If the software could look ahead and detect the OH and then ramp up the fan before the overhang I think overhangs with PET-G or ABS will be awesome.

levleon13, 2022-09-19:
> "Why not have a warmup value? So how many ms the fan is to "warmup" and add that in advance so the fan is up to speed before it reaches the overhang?"
>
> "Is this something that might be added to [...]

This is already possible :)
1. overhangs_width_speed - change % to the desired detection.
2. Bridge fan speed - select desired fan on overhangs. 
DONE.

Garryb123, 2022-09-19:
> This is already possible :)
> 
> 1. overhangs_width_speed - change % to the desired detection.
> 2. Bridge fan speed - select desired fan on overhangs.
>    DONE.

I'll try; thank you!