supermerill/SuperSlicer #415: Generating Gcode can be verry slow

Stephan3, 2020-08-19:
### Version
latest from release tab -> 2.2.52.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Klipper latest Master

### Behavior
"generating gcode" takes for ever. it easily can take 30 minutes on my mechanical part plates on a VM.
The VM hostsystem has sufficent resources Xeon 32c/64t
It feels like it often takes place when i slice the plate for printer_a and switch to printer_b to let it do the same plate.

#### Project File (.3MF) where problem occurs
attached
[plate_002.zip](https://github.com/supermerill/SuperSlicer/files/5099102/plate_002.zip)

[image]

supermerill, 2020-08-19:
yeah, it's mono-thread.
making it more efficient is a big task. I don't have the time/money for that right now.

Stephan3, 2020-08-20:
I just wondered that Kisslicer was 10 times faster. Maybe i am doing something wrong here? Are there tips of things to avoid to make it faster? Maybe slice per opject  and not per layer? each object one thread or whatever.