supermerill/SuperSlicer #416: Seam-Sphere does not work sometimes.

JoduWei, 2020-08-20:
### Version
f741e280310df9b98638db684208fb429fca4bd5

### Operating system type + version
Win 64

### Behavior
Seam Placement does not work always.

[image]
[image]
[seam_placement.zip](https://github.com/supermerill/SuperSlicer/files/5103488/seam_placement.zip)

supermerill, 2020-08-20:
I'll explain a bit more how it works

- for each object (here, you have only one)
- for each layer
- it get the sphere with the center point z-coordinates nearest of the z position of the layer. It doesn't use xy coordinates at all (and as it use the center, stretching it doesn't do anything)
- it will put the seam at the nearest point, taking only into account the xy position of the center of the sphere.

I can change the behavior.

JoduWei, 2020-08-20:
Thanks for your reply.
I'm using plater to create these plates. There is no option to create 3fm/amf files unfortunately.
Combinig Plater and SSlicer would be great idea. :D 

> it get the sphere with the center point z-coordinates nearest of the z position of the layer. It doesn't use xy coordinates at all (and as it use the center, stretching it doesn't do anything)
Okay understood. It's only the direction. It's like Kisslicer but always visible. An improvement. :)

https://github.com/Rhoban/Plater

JoduWei, 2020-08-20:
I thought it would work like this example. It would hide all seams perfectly. :D 
[image]

supermerill, 2020-08-20:
I'll change it to:
 - if sphere: nearest xyz
 - if cylinder: nearest xy, if it's inside the height of the cylinder.

JoduWei, 2020-08-20:
Great. Ty. :)

closed 2020-10-22 by supermerill