supermerill/SuperSlicer #420: Getting back upstream

lordofhyphens, 2020-08-23:
@supermerill what needs to happen on my end to get most of this PR'd back up to slic3r/Slic3r?

supermerill, 2020-08-24:
hi

>@supermerill what needs to happen on my end to get most of this PR'd back up to slic3r/Slic3r?

For this post i'm assuming by "this" you mean "this repository". If you talk about a specific pr, well i'm out of the loop.

I'm "surfing the prusa waive", in a sense that the first feature is to merge their changes. That said, most change they do is also here. I'm doing that because they have much more resources, and so I can't "compete" with them (and i don't want to), so it's better to work with.

Also I'm growing this fork to be feature creep. I'm not sure it's your goal. Instead of getting "most" of the feature, maybe it's best to concentrate the effort on "best" or "must have".

I'm trying to create consistent commit, so it's easier to cherry pick in case someone want to merge them with an other branch. So I'm going to talk about commits.

#### commits that add a setting and the only changes are inside libslic3r:

these may be mostly straiforward, but there are some thing to consider:
* settings are not added the same way
   * in prusaslicer, you have to add the entry in printconfig.cpp / hpp, like in slic3r, the string def in preset.cpp (gui) and the entry in tab.cpp
   * in superslicer, you have  to add the entry in printconfig.cpp / hpp, like in slic3r, the string def in preset.cpp (gui) but I've moved (most of) the tab def into ui files, so everyone can create their dream setting gui.
 * every Point / Vec is now using Eigen, hence the x() instead of x
 * Prusa added some more functions to utility class (like explolygon) and deleted some other functions. So some functions have to be ported back into mainline, or the code need to change.

Commits on infill, perimeters, brim, skirts (and maybe supports) should be straightforward.

#### commits that modify gcode:

from what I remember, they are very different so it may be hard to merge.

#### commits that modify the gui (but not the settings

I won't recommend merging them, at least i won't.
Calibration & Freecad dialog may sound more m"mergeable" as they seems "separate", but they are wxwidget cpp code that interact with the cpp gui class for refresh & grabbing the settings. I don't know how it can be merged into perl without rewriting it in perl.

### notes

> what needs to happen on my end to get most of this PR'd back up to slic3r/Slic3r?

By me? A lot.
A subset of the features can be merged more easily, but they will clutter the setting gui a lot, and i don't know if you want it.

Some tough: From what i remember, it's a bit hard to add a new setting, and the number of modified files should be reduced to the maximum. But it's only useful if you accept to have a bazillion of settings.

Also my test coverage is not optimal.

lordofhyphens, 2020-08-24:
Unless Prusa's devs have spent a lot of time writing tests, I suspect your test coverage is a lot better than theirs :D

lordofhyphens, 2020-08-24:
FYI I added you to a team that can add branches, commit to slic3r/Slic3r.

yschroeder, 2020-08-29:
Maybe I am highjacking this issue a bit, as the relevant people are currently around, I ask here anyway:

What is the project status of Slic3r? I have no idea what happened to Slic3r since Prusa hard forked. There are no releases, no changelogs, but there is a steady stream of commits on the master branch, the project looks kind of alive. Are you planning new releases? Is it supposed to be rolling release? Are you still trying to backport Prusa stuff? I remember a time when Prusa still called it SlicerPE, most of the work on Slic3r was backporting. 

I am sharing the opinion with Merill, that you can only surf the Prusa wave and take their version as a base and apply new cool features on top, hoping they will integrate it (or creating PRs). They have much more manpower, you cannot keep up with their speed. But I also understand that Slic3r does not want to surrender to Prusa Slicer. 

Basically, SuperSlicer works as a project, because Prusa Slicer is very popular and running SuperSlicer is like a new expert mode you didn't know before with lots of options to play with (many of them are not substantial for good prints, but fun to experiment nonetheless). And Merill is very open to discuss and implement niche features that might help in very specific situations. Also, Merill does not fear to touch the slicing core, Prusa seems to avoid that to not introduce regressions. Most of their changes are GUI and configuration stuff. 

Personally, I wouldn't have bothered to test SuperSlicer if it was a completely new slicer, what it sold for me was that it only wants to be a Prusa Slicer on steroids.

I think my main problem is that I don't even know what I am missing out by not using Slic3r (anymore) as my only way to figure out what new cool features it has is to read commit messages.

And in the end all this text might only be a feature request for Slic3r: Make releases with changelogs.

lordofhyphens, 2020-08-29:
@yschroeder technically active but slow. I have been slowly reviewing the Prusa fork to see how it would be for rebasing Slic3r on it. My last experience with their code was a couple years ago.

Prusa has people that he pays to work on their fork for their products, but not to push code back upstream. I have a full time job that has nothing to do with 3d printing or Slic3r.

I get more feature requests than pull requests and that has been the state of it for basically forever.

n8bot, 2020-08-29:
Let's be blunt: what parts of PrusaSlicer are you *against* merging with upstream slic3r? Hasn't Prusa done a ton of work porting things to c++ code? They've adopted the Eigen library in favour of custom types, etc. What is it that you'd like and what is it that you wouldn't like?

I'm a relative newb when it comes to software dev and c++, but I have time to focus on this because 3d printing *is* my full time job. I don't think I'll be very useful writing code or anything, but I have spent the past few months getting a little bit acquainted with the PS codebase, and I'm getting to the point where I can sort out what's what, why's why, and where's... where(?).

Once thing I have noticed is that upstream Slic3r's viewport is not quite as responsive as the one in PS. So, the "usability" for more detailed models suffers slightly. Is this because of the port to c++ they did? That might be a good place to start incorporating the PS changes.

Ss for the libslic3r core, it does seem in some ways PS devs are leaving it alone, but in other ways they're very much messing about with it. I think the core needs to be streamlined and refactored for the new decade. It's starting to show its age and some wear (worn away by the dozens of fingers that have been hacking away at it).

Let me know if my comments are unwelcome and off-base. I'll kindly eff-off if you want.

closed 2021-06-04 by supermerill