supermerill/SuperSlicer #422: Feature request: fixed fan speed settings for each section of the printed object

robustini, 2020-08-25:
It would be interesting to be able to set the fan speed for each individual printed section, for example:

Default fan speed: 30%
Bottom layers: 50%
Top Layers: 100%
External perimeters: 50%
Internal perimeters: 20%
Infill: 50%
Bridge: 100%
Supports: 100%

It seems to me that this is currently not possible.

supermerill, 2020-08-25:
What is the use-case?

robustini, 2020-08-25:
The case is precisely what is written above.
We may need to have a higher separation of fan settings, currently it is not possible to do the above.
"All other extrusions" in my opinion it is too generic for some printed object.
My two cents...

robustini, 2020-08-25:
For example these are my settings, and I don't understand why the fan is running at 34% during the infill.
If there was a specific value for the infill it wouldn't happen.

[image]

supermerill, 2020-08-25:
> I don't understand why the fan is running at 34% during the infill.

> Attiva centola se la stampa del layer impiega meno di 60 secondi

If you read the tootip, it says that it will increase the fan speed from the base speed set below (for infill, it's the default at 1%) to the top speed set at 100% (the field just below

robustini, 2020-08-25:
In fact the percentage modulation can be convenient, but personally I would have preferred separate settings declared, for each section.
If it is not possible to implement it no problem, mine was just an idea perhaps useful in some cases.
Thanks anyway!

supermerill, 2020-08-25:
You can disable the percentage modulation for short layer.

closed 2020-10-18 by robustini