supermerill/SuperSlicer #423: Crash application (2.2.53.0).

Shama71, 2020-08-25:
### Version
2.2.53.0
### Operating system type + version
Win64
### Behavior
Crash application.

#### Project File (.3MF) where problem occurs

[flow_test_cube_v2_for_0.4_nozzle.zip](https://github.com/supermerill/SuperSlicer/files/5123578/flow_test_cube_v2_for_0.4_nozzle.zip)

neophyl, 2020-08-25:
Same version of Super Slicer (2.2.53.0 on Windows 10 64 bit).
Project file opens but on attempting to slice application shuts down with no visible error messages.
Event viewer in windows only has 
 superslicer.exe 
   2.2.53.0 
   5f4046d2 
   slic3r.dll 
   0.0.0.0 
   5f4045af 
   c0000005 
   00000000001451fe 
   ac7c 
   01d67ad83ccecc5c 
   G:\SuperSlicer_2.2.53.0_win64_20200823\superslicer.exe 
   G:\SuperSlicer_2.2.53.0_win64_20200823\slic3r.dll 
   2e6dd9b0-8ff7-4bf3-b3bc-bf0d6cf2edad 

Changing to my normal print/filament and printer profiles and the model slices.
Changed Printer back to the OP printer and sliced, -sliced ok
Changed filament back to the OP filament and sliced - sliced ok
Changed Print settings back to OP print settings and sliced - crash again so something about the print settings profile it doesnt like.

neophyl, 2020-08-25:
Ok found the issue.  Brim width is set to 1mm and Brim Offset is also set to 1mm.  The tooltip for brim offset mentions that the offset value is 'substracted' (which I believe should be" subtracted") from brim width or interior brim width so must be lower than those values.  If set to 0.5mm for example the project will slice correctly.  

Probably want to put a check in there to make sure the value is always less.

supermerill, 2020-08-25:
gotcha

Shama71, 2020-08-25:
This settings from previous version.
Previous version not crash.

supermerill, 2020-08-25:
The brim in the previous version can be buggy.
It has been rewritten.
I'm also adding a safeguard/fix (currently, it crash if i ask him to extrude 0 extrusion)

closed 2020-09-03 by Shama71