supermerill/SuperSlicer #424: SuperSlicer does not run on Mac OSX with case sensitive filesystem

mesfet, 2020-08-26:
### Version
2.2.53

### Operating system type + version
MacOS 10.13.6

### Problem description
SuperSlicer does not run after installation on MacOS. The program icon in the Finder shows a generic executable icon.  Executing the program does not produce and application windows or errors.

### Steps reproduce
Install MacOS version on case sensitive file system and try to run the program. 

### Workaround
View the package contents of the SuperSlicer application.  Under the Contents folder rename the "resources" folder to "Resources"

mesfet, 2020-09-02:
I downloaded and tested SuperSlicer 2.2.53 and the problem has been fixed for me.

supermerill, 2020-09-02:
hurrah!

closed 2020-09-18 by mesfet

BobB84, 2023-09-24:
HI, where can I find this MacOS version now? Is it still available? When searching for it all I can find is linux and Win64 options