supermerill/SuperSlicer #425: Improved G10 Implementation

lukeslaboratory, 2020-08-26:
### Version
2.2.53

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Duet Machines, Particularly Toolchangers

### Behavior
In the changelogs, if you have RRF selected as your firmware, the slicer outputs G10 commands to set temperatures instead of M commands - Great!

Can there be some things ironed out?

 When using a single tool on my toolchanger (tool 0 in particular) I have to go back in and edit the G10 command to indicate the correct tool, as using G10 S215 gives an error and fails to set temperature. I'd prefer a verbose tool setting.

The other half of this is being able to set a standby temperature for toolchangers - There's already a multimaterial setting for enabling toolchange temperatures, I'd like to have this used to set the R value as the other half of the G10 temperature command.

Thanks!

supermerill, 2020-08-26:
> Can there be some things ironed out?

of course!

> as using G10 S215 gives an error and fails to set temperature. I'd prefer a verbose tool setting.

what I need to write in which circumstances?

>  I'd like to have this used to set the R value as the other half of the G10 temperature command.

What do you mean? Can you be more verbose?

lukeslaboratory, 2020-08-26:
> what I need to write in which circumstances?
When using a single tool, especially tool 0. 

From what I can tell, if the tool selected is any tool other than tool 0, the slicer adds in the correct tool selection, but for T0, they're omitted. I think the behavior should be to write the tool selected in every G10 command. Instead of starting a single-tool print with just G10 S215, it should be G10 P0 S215, and continues referencing T0 for the rest of the file.

> 

> I'd like to have this used to set the R value as the other half of the G10 temperature command.
> What do you mean? Can you be more verbose?

The other half of the G10 command for setting temperatures is the standby temp, which uses Rnnn to set a temperature for when the tool is placed on standby. G10 P0 S215 R200 would use a printing temp of 215C, and when the machine goes to drop off T0 and Pick up T2, T0 would then start holding 200C and T2 would start using its active temp. This behavior is very appreciated with any finnicky filaments that tend to clog if left cooking too long. 

Overall  - I believe ideal starting gcode automatically inserted should be:

```
;Temperature settings for whatever tools are used in the print
G10 P0 S215 S200 ; Tool 0, PLA
G10 P1 S230 S200 ; Tool 1, ABS
G10 P2 S200 S150 ; Tool 2, PVA

T{first tool} ; Turn on first tool (this also isn't used on single-tool prints when the tool is T0, but is needed for multitool machines)
***rest of print gcode***
```

supermerill, 2020-08-26:
> standby temp

new setting for a filament?

lukeslaboratory, 2020-08-26:
To the best of my knowledge, standby temps have been part of RRF's implementation of G10 for quite some time now, at least RRF 2.0 [https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting](url)
I was thinking that it would be nice to use the "Toolchange temperature" in the multimaterial section for the standby value, if enabled. That's the first thing I tried when I saw that G10 was implemented.

I think for calling out T0, it may not break anything if its omitted on a single-extruder machine, but for a multitool machine it's needed.

supermerill, 2020-08-26:
> It may be more prudent to select the active tool (guaranteeing that the tool is in "active" state, and then issue a M116 command, even for single-tool machines.

If it's a single-tool printer, and it needs to have a tool selected, it's not a standard config imo (no firmware need it by default, right?). Here, the selection of the tool should be set by the user in the starting gcode.
That said, if rrf really need a tool selection in all cases in the startup, i'll add it.

todo:
* use Toolchange temperature as R setting for g10
* put a G10 for all extruders at the start to set the R, like `G10 P0 R180 S215`
* ensure T0  is called if multi-tool printer, even if only one filament is used
* write G10 P0 if only one extruder. Have to always write P or it's a retract command.

it's right?

lukeslaboratory, 2020-08-26:
> If it's a single-tool printer, and it needs to have a tool selected, it's not a standard config imo (no firmware need it by default, right?). Here, the selection of the tool should be set by the user in the starting gcode.
That said, if rrf really need a tool selection in all cases in the startup, i'll add it.

I just tried G10 S230 on my single-tool machine running RRF3 - it gave me an error as the tool # was not supplied. I then ran G10 P0 S230, and the active temperature updated, but the tool did not start heating, as it remained off.  I then entered M116, but nothing happens, as it appears that M116 only affects tools that are not "off" for wait times. 

I went over and read documentation for M109/M104, it looks like not selecting a tool wouldn't have been noticed on single-tool machines since if no tool is supplied, M109 selects and activates the lowest-indexed tool, meaning all future extrusion commands would use that tool. Using G10 doesn't invoke that behavior, necessitating tool activation with a T command before printing. 

In summary, your list looks good, but I believe that you'll also need to always have an explicit tool command if using **only** G10 commands, regardless of single or multi tool, because G10 does not activate tools for heating/use.

supermerill, 2020-08-27:
When The printer is a multi-extruder but with one hot-end, what does it need?
same as with n hotend?

lukeslaboratory, 2020-08-28:
From what I can tell, there is a difference between mixing and non-mixing variants. Non-mixing will just require seperate G10 settings for temperatures and will still need T values called out to switch between temperatures/heads. For mixing hotends, it will be defined as a single tool (which will still need to be selected at print startup) with a single temperature, and will use either Multiple E values, or will have a Mix ratio for extrusion.

From the dozuki on duet's gcode setup:

> It is permissible for different tools to share some (or all) of their drives and heaters. So, for example, you can define two tools with identical hardware, but that just operate at different temperatures.

supermerill, 2020-09-06:
can you test the [nightly version](https://github.com/supermerill/SuperSlicer/actions)?

lukeslaboratory, 2020-09-07:
Hi There - it took longer than I'd like to admit to find the nightly build.

I did a 5-tool test slice to see what the outputs are. Below is my starting gcode generated.
```
M107
M190 S110 ; set bed temperature and wait for it to be reached
G10 P0 R200 S260 ; sets the standby temperature
G10 P1 R200 S250 ; sets the standby temperature
G10 P2 R185 S240 ; sets the standby temperature
G10 P3 R200 S230 ; sets the standby temperature
G10 P4 R200 S250 ; sets the standby temperature
T0
G92 E0
G10 P0 S265 ; set temperature
G10 P1 S250 ; set temperature
G10 P2 S240 ; set temperature
G10 P3 S230 ; set temperature
G10 P4 S250 ; set temperature
G28 ; first line of starting gcode I defined in machine profile
G32 ; auto-magic levelling
M558 H3 F240 A1
M557 X50:50 Y100:100 S20 ; to be later updated by auto-mesh-sizing-scripts
G29 ;start of starting gcode
M572 D0 S0.01; Set K-factor PLA
M572 D2 S0.01; Set K-factor PLA
M572 D2 S0.022; Set K-factor PLA
;M572 D3 S0.01; Set K-factor PLA
;M572 D4 S0.01; Set K-factor PLA
;M572 D5 S0.01; Set K-factor PLA
M106 P15 S255;  TURN ON THE LIGHTS
M117 We Prantin
G10 P0 S265 ; set temperature
M116 ; wait for temperature to be reached
G10 P1 S250 ; set temperature
M116 ; wait for temperature to be reached
G10 P2 S240 ; set temperature
M116 ; wait for temperature to be reached
G10 P3 S230 ; set temperature
M116 ; wait for temperature to be reached
G10 P4 S250 ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
```

The initial start block is perfect - sets up all of the Tools with their corresponding Active and standby temps - Great! 

It then auto-selected the first tool. This in rrf-land will then go execute any macros associated with that tool. if its a single-tool machine, there probably aren't any, and this is fine. However- on a multi tool system that uses any macros with motion defined, this will cause it to attempt to pick up the tool, do a dance, whatever's programmed into the macro. Since preheats come before homing (or any of the start gcode), this will cause it to think that it has a tool grabbed before homing, throwing all of the remaining tool offsets off whack relative to 0. Second, It only activates the first tool - the 4 remaining tools remain in "off" position, doing nothing. 

Proposal - After defining temps using the G10 commands, do a cycle of Tn P0 commands followed by a T-1 P0. This will activate the tools and at the end will result in all of them being heated to their standby temperatures, but without causing any machine motion or applying tool offsets. After this point, the normal startup script will activate. (mine starts with G28)

Next, it selects the first layer temperatures. It uses the correct Syntax, which is great! I'd suggest that this section be omitted and just added to the initial G10 commands, and when the first layer is complete, just use G10 Snnn to update the active temperature on the tools.

Following the M117, it then uses another round of G10 temp setting, followed by M116 to wait for heatup. This doesn't work as intended, for a few reasons - G10 only sets the Active and standby temperatures, and those are permanently set, meaning they do not need to be commanded if not changed between toolchanges. If the tool is in the Active state, it will be commanded to meet its active temperature. If the tool is in standby, it will be commanded to meet its standby temperature. if the tool is off, it will not be included in M116 commands. When switching between Tn to Tx, Tn will be put in standby, and Tx will be put on its active temperature.

In this initial slice, only T0 has been commanded outside of its "off" state, and will be the only tool checked for M116 compliance. The other 4 tools will not be checked since they are still off. In order to use M116 in this way, the tools would need to be activated per check (Tn before M116), and will NOT need a G10 command, since we have specified the temperature states before.

That being said - M116 will very likely be in the toolchange macros for tool pickup (e3d's default) and is not very valuable for any tool but the active tool. I don't think that it belongs unless its a single-tool machine.

Here's how I have to edit my starting Gcode to make it function properly:

```
M107
M190 S110 ; set bed temperature and wait for it to be reached - Ideally this would be a M140, with M190 following the T-1 P0 command
G10 P0 R200 S265 ; sets the standby temperature
G10 P1 R200 S250 ; sets the standby temperature
G10 P2 R185 S240 ; sets the standby temperature
G10 P3 R200 S230 ; sets the standby temperature
G10 P4 R200 S250 ; sets the standby temperature
T0 P0
T1 P0
T2 P0
T3 P0
T4 P0
T5 P0
T-1 P0 ; After this point, all extruders will be in the "standby" state and will be heated to the value defined by Rnnn. No Macros will be run
G28 ; first line of starting gcode I defined in machine profile.
G32 ; auto-magic levelling
M558 H3 F240 A1
M557 X50:50 Y100:100 S20 ; to be later updated by auto-mesh-sizing-scripts
G29
M572 D0 S0.01; Set K-factor PLA
M572 D2 S0.01; Set K-factor PLA
M572 D2 S0.022; Set K-factor PLA
;M572 D3 S0.01; Set K-factor PLA
;M572 D4 S0.01; Set K-factor PLA
;M572 D5 S0.01; Set K-factor PLA
M106 P15 S255;  TURN ON THE LIGHTS
M117 We Prantin
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
T0 ; select first tool used, will very likely have a M116 command in the toolchange macro, but M116 following would ensure temperatures are correct.
G92 E0
```
Hopefully this helps!

closed 2020-10-22 by supermerill

guildner, 2020-11-19:
I seem to be hitting this issue on 2.2.53.4 (latest stable as of writing this comment)

When slicing for a toolchanging printer, but using only a single tool (T0) for a particular print, the tool does not get selected or picked up at the start of the print.

Should this be resolved in the stable build?

SputnikOC3D, 2021-09-04:
Any headway on this issue ?  Ver 2.3.56.8 here and same challenges.

SSlicer issuing G10 S0 S215 and a print killing M116 appended to my custom start.g code in both reprap/srpinter flavor and reprap flavor selected.

Additionally it should be noted to the devs that ReRap FW 3+ indicates G10 for tool temp settings is deprecated and use of M568 is now preferred.

One should not have to stuff fake M104 / M190 at the front of their start.g custom code [ especially if one is calling a reprap MACRO to start their print ... ] to trick the slicer software from injecting the above code.

Im running a 1LC toolboard on voton and the "safeguard" code is halting my print at full temps / print height at the end of my purge line.  When I go in and manually extricate the garbage out [ 2 lines of spurious auto generated code, print fires off as expected.

Ive seen posts about how to stop the slicer from auto generating any code but cannot locate the OFF Switch or setting - please advise

supermerill, 2021-09-04:
> Additionally it should be noted to the devs that ReRap FW 3+ indicates G10 for tool temp settings is deprecated and use of M568 is now preferred.

 ... again?
 
 > Ive seen posts about how to stop the slicer from auto generating any code but cannot locate the OFF Switch or setting - please advise

[image]

SputnikOC3D, 2021-09-04:
Many thanks for assisting the blind ... sorry.