supermerill/SuperSlicer #426: Unwanted "supports" inside the walls

robustini, 2020-08-26:
I don't find the option to remove these supports (in the firt image) inside the perimeters which are generated only if the infill is present.
If I put infill at 0% this addition disappears (second image), yet I hink there is an option in the "Infill" section to not generate them, or is missing?

SuperSlicer V2.2.53

[image]

[image]

[FUSO1.zip](https://github.com/supermerill/SuperSlicer/files/5129573/FUSO1.zip)

neophyl, 2020-08-26:
You may want to read this issue in the main Prusa Slicer repository https://github.com/prusa3d/PrusaSlicer/issues/3744

Pavulon87, 2020-08-26:
> You may want to read this issue in the main Prusa Slicer repository [prusa3d#3744](https://github.com/prusa3d/PrusaSlicer/issues/3744)
> or this one here #253 (took me awhile to find it)

I just lost one hour looking for ensure_vertical_shell_thickness in code to find this post :D
Layer modifier is great work-a-round.

robustini, 2020-08-26:
I tried the workaround but it produces the same result here.

[image]

Pavulon87, 2020-08-26:
@robustini, are you sure that "Ensure vertical shell thickness" is turned off?

supermerill, 2020-08-26:
If ensure_vertical_shell_thikness is deactivated, a little codepath (printobject line 3200) change sparse infill to solid infill if the gap is < 3 * solid extrusion width. This is deactivated if fill_density is 0.

that's the problem here

supermerill, 2020-08-26:
For now, if you enable "ensure vertiacl thickness", you have much less solid infill. But no zero.

Pavulon87, 2020-08-26:
> note that there is this comment before this code block:
> //FIXME Vojtech: Disable this and you will be sorry.
> 
> edit: disabling this doesn't seems that problematic.
Maybe it will be a worse solution for some models, but I believe that commenting on it gives a better result.
Much less unnecessary printing after commenting. And without visible problems when the option is activated.

robustini, 2020-08-28:
> 
> 
> @robustini, are you sure that "Ensure vertical shell thickness" is turned off?

Open my project in the zip file and verify, is disabled but the problem is still there.

haleypearse, 2020-08-29:
Yes I have the same thought. For two years I've checked every release to see if anyone's bothered to fix this absurd bug. You're right it seems like it would be simple. I believe is written in Perl. Although I'm basically a layperson I'm tempted to have a try.

RolfLindbom, 2021-03-27:
[image]
[image]

After some very layman-ish fiddling (I'm an indie Unity gameplay coder...) I found that disabling line 2505 in PrintObject.cpp (PrusaSlicer v2.4 alpha) seems to produce the desired results when "Ensure vertical shell thickness" is deselected.