supermerill/SuperSlicer #428: Incorrect slicing, might be related to holes.

kursatu, 2020-08-27:
### Version
_Version of SupserSlicer used goes here_
2.5.53. I am build the one from the latest sources on GitHub.
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
>git describe --tag
2.2.53.0
### Operating system type + version
_What OS are you using, and state any version #s_
[image]

Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Custom CoreXy Vking derivative
Marlin bugfix-2, latest code
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
I think it started to happen with 2.5.53 because I've printed these before.
A hole is filled for some reason which ruins the whole print.
There are no reported manifold errors in the model.
* _Steps needed to reproduce the problem_
Use the attached .3mf file and slice
Observe that one os the holes is filled for some reason.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_

* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File ->
[image]
[image]
[Defect5.zip](https://github.com/supermerill/SuperSlicer/files/5138151/Defect5.zip)

 Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-08-27:
[image]

Can't reproduce. Maybe try with the very new release...

Also, be careful, you set the left wing on the wrong side.

kursatu, 2020-08-27:
I've pulled the latest sources and built it.  The problem is not there anymore.
Thanks for looking into it.

closed 2020-08-27 by kursatu