supermerill/SuperSlicer #429: add options for z-hop function

belm0, 2020-08-29:
# Behavior
* _Describe the problem_
z-hop of slicer is not flexible (vertical hop only).  There are conflicting requirements (reduce blobbing, avoid hitting existing extrusions) which cannot be addressed by this simple function.
* _Expected Results_
add control over z-hop function (raise it during wipe according to non-linear function, etc.)

promising results here: https://github.com/prusa3d/PrusaSlicer/issues/2507#issuecomment-682599025

_Is this a new feature request?_
yes

supermerill, 2020-08-29:
please explain more.
"etc" is not enough for me to implement it.

n8bot, 2020-08-29:
I just added a comment to that PS issue, talking about how IceSL handles their zhop/travel moves. They do it diagonally, which has the beneficial side-effect of slowing down nozzle speed just before "unlifting," which helps to alleviate ringing that can sometimes occur in the pritn move that directly follows a fast travel move.

closed 2021-11-04 by supermerill