supermerill/SuperSlicer #430: Never switches to second layer extrusion temperature

ManuelMcLure, 2020-08-29:
### Version
SuperSlicer_2.2.53_2020-08-27

### Operating system type + version
Windows 10 2004

### 3D printer brand / version + firmware version (if known)
Anet AM8 BLV MGN12 Mod, Marlin Bugfix-2.0.x on Re-ARM/RAMPS.

### Behavior
* Generated gcode does not switch from the first layer extrusion temperature to the subsequent layer temperature
* In filament settings, set first layer temperature to 210 degrees and following temperature to 200 degrees. Generated gcode will not include any ``M104 S200`` - instead I see the following:
```
; LAYER: 1, Z HEIGHT: 0.4
G10 S200 ; set temperature
M140 S60 ; set bed temperature
```
It looks like it's generating a ``G10`` instead of an ``M104``.

#### Project File (.3MF) where problem occurs
[Vesa75x75Mount.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5145905/Vesa75x75Mount.3mf.zip)

ManuelMcLure, 2020-08-29:
If I change the Gcode flavor to Marlin instead of RepRap the problem does not occur. I had it set to RepRap to avoid generating machine limits when I was using PrusaSlicer and that setting was propagated to SuperSlicer. It still seems like it should not be generating a ``G10`` even in RepRap mode.

ManuelMcLure, 2020-08-29:
OK, I see that ``G10`` is a valid way to set temperature on RepRap Firmware. So this might not be a bug. I'll close it since SuperSlicer has a way to avoid putting machine limits in the gcode even with Marlin flavor.

closed 2020-08-29 by ManuelMcLure

supermerill, 2020-08-29:
To avoid printing machine limits, just uncheck the option.