supermerill/SuperSlicer #431: Fan speed bug with MMU

hagan66, 2020-08-30:
### Version
2.2.53.1

### Operating system type + version
MacOS 10.15.6

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5S FW 3.9.0

### Behavior
Fan speed gets messed up with MMU prints. For example I set up the simplest 100% fan speed in filament settings. But when I inspect the preview the fan drops to 50% at the first filament change. If I set up different fan speed for infill, perimeters etc, it looks even worse.

[image]

#### Project File (.3MF) where problem occurs

[Kub1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5146487/Kub1.3mf.zip)

hagan66, 2020-09-02:
This is really bad actually. I havent been able to find a workaround and it prevents me from printing multi material prints. And I love Superslicer too much to go back to Prusaslicer.

supermerill, 2020-09-07:
that wipe tower.... one day, it needs to be nuked and redone with the gcode writer.

closed 2020-10-22 by supermerill