supermerill/SuperSlicer #432: 2.2.53.1 build fails on NixOS

cab404, 2020-08-31:
### Version
2.2.53.1

### Operating system type + version
NixOS 20.03, latest nixpkgs

### Behavior
I am trying to package SuperSlicer for NixOS, and can't seem to build the thing. I'm using the same environment as for PrusaSlicer — that means

```
$ make -j16
[  1%] Built target encoding-check
[  1%] Built target hidapi
[  2%] Built target avrdude-conf-gen
[  3%] Built target clipper
[  3%] Built target miniz_static
[  4%] Built target nowide
[  6%] Built target polypartition
[  8%] Built target poly2tri
[  9%] Built target glu-libtess
[ 13%] Built target admesh
[ 19%] Built target qhullstatic
[ 20%] Built target imgui
[ 20%] Checking source files encodings for target avrdude
[ 20%] Checking source files encodings for target libslic3r_gui
[ 21%] Checking source files encodings for target libslic3r
[ 21%] Built target libslic3r_cgal
[ 23%] Built target libnest2d
[ 23%] Checking source files encodings for target avrdude-slic3r
[ 23%] Checking source files encodings for target semver
[ 25%] Built target Shiny
[ 25%] Built target gen_conf_h
[ 25%] Built target encoding-check-avrdude-slic3r
[ 25%] Built target encoding-check-semver
[ 25%] Built target libnest2d_tests
[ 25%] Built target encoding-check-avrdude
[ 26%] Built target encoding-check-libslic3r_gui
[ 26%] Built target encoding-check-libslic3r
[ 26%] Built target global-encoding-check
[ 27%] Built target semver
[ 36%] Built target avrdude
[ 36%] Built target avrdude-slic3r
[ 63%] Built target libslic3r
[ 64%] Building CXX object tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_data.cpp.o
[ 64%] Building CXX object tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_complete_objects.cpp.o
[ 64%] Building CXX object tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_print.cpp.o
[ 65%] Built target sla_print_tests
[ 69%] Built target libslic3r_tests
[ 73%] Built target fff_print_tests
[ 97%] Built target libslic3r_gui
[ 98%] Built target slic3r
[ 98%] Built target slic3rutils_tests
In file included from /home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.cpp:1:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp: In function ‘bool Slic3r::Test::_equiv(const T&, const T&)’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp:61:77: error: ‘epsilon’ is not a member of ‘Slic3r::Geometry’
   61 | bool _equiv(const T& a, const T& b) { return abs(a - b) < Slic3r::Geometry::epsilon; }
      |                                                                             ^~~~~~~
In file included from /home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_print.cpp:7:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp: In function ‘bool Slic3r::Test::_equiv(const T&, const T&)’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp:61:77: error: ‘epsilon’ is not a member of ‘Slic3r::Geometry’
   61 | bool _equiv(const T& a, const T& b) { return abs(a - b) < Slic3r::Geometry::epsilon; }
      |                                                                             ^~~~~~~
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.cpp: In function ‘void Slic3r::Test::init_print(Slic3r::Print&, std::initializer_list<Slic3r::Test::TestMesh>, Slic3r::Model&, Slic3r::DynamicPrintConfig*, bool)’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.cpp:269:76: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  269 |  DynamicPrintConfig &config = Slic3r::DynamicPrintConfig::full_print_config();
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.cpp: In function ‘void Slic3r::Test::init_print(Slic3r::Print&, std::vector<Slic3r::TriangleMesh>, Slic3r::Model&, Slic3r::DynamicPrintConfig*, bool)’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.cpp:304:76: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  304 |  DynamicPrintConfig &config = Slic3r::DynamicPrintConfig::full_print_config();
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
[... 18 lines cut ...]
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_print.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____36()’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_print.cpp:215:83: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  215 |         DynamicPrintConfig& config = Slic3r::DynamicPrintConfig::full_print_config();
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_complete_objects.cpp:7:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp: In function ‘bool Slic3r::Test::_equiv(const T&, const T&)’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_data.hpp:61:77: error: ‘epsilon’ is not a member of ‘Slic3r::Geometry’
   61 | bool _equiv(const T& a, const T& b) { return abs(a - b) < Slic3r::Geometry::epsilon; }
      |                                                                             ^~~~~~~
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_complete_objects.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____0()’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_complete_objects.cpp:53:83: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
   53 |         DynamicPrintConfig& config = Slic3r::DynamicPrintConfig::full_print_config();
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_complete_objects.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____15()’:
/home/cab/data/cab/nixpkgs/superslicer/source/tests/superslicerlibslic3r/test_complete_objects.cpp:127:83: error: cannot bind non-const lvalue reference of type ‘Slic3r::DynamicPrintConfig&’ to an rvalue of type ‘Slic3r::DynamicPrintConfig’
  127 |         DynamicPrintConfig& config = Slic3r::DynamicPrintConfig::full_print_config();
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[2]: *** [tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/build.make:121: tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_data.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/build.make:108: tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_print.cpp.o] Error 1
make[2]: *** [tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/build.make:95: tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/test_complete_objects.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1604: tests/superslicerlibslic3r/CMakeFiles/superslicerlibslic3r_tests.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
```

cab404, 2020-08-31:
CMake configuration:
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/6is25fyx29d731idycngl7qmgcax5xng-gcc-wrapper-9.3.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/6is25fyx29d731idycngl7qmgcax5xng-gcc-wrapper-9.3.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: 1
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: OFF
```
Path is huge due to Nix build system, but basically contains following libraries:
```
    boost
    cereal
    cgal_5
    curl
    eigen
    expat
    glew
    gmp
    ilmbase
    libpng
    mpfr
    nlopt
    openvdb
    systemd
    tbb
    wxGTK31
    xorg.libX11
```

supermerill, 2020-08-31:
you can build with  -- SLIC3R_BUILD_TESTS: OFF

closed 2020-08-31 by cab404

cab404, 2020-09-01:
[Here's a PR for Nixpkgs](https://github.com/NixOS/nixpkgs/pull/96882), if you are interested)

supermerill, 2020-09-01:
there's a "slic3r++" in this pr... i wonder if i forgot to replace this one...

cab404, 2020-09-01:
you probably did)
It won't be merged for some time – Nixpkgs is like that if you don't ping some people – so if you fix it, I would fix it in PR)

cab404, 2020-09-01:
> there's a "slic3r++" in this pr... i wonder if i forgot to replace this one...

https://github.com/supermerill/SuperSlicer/compare/master...cab404:patch-1