supermerill/SuperSlicer #433: Preferences -> Use perspective camera, lose changes

Rodrigormaraujo, 2020-08-31:
### Version
2.2.53

### Operating system type + version
Windows 10 PRO

### 3D printer brand / version + firmware version (if known)
Hypercube with RepRapFirmware for Duet 2 WiFi/Ethernet 2.05 (2019-12-13b1)

### Behavior
In Preferences dialog, option ' Use perspective camera' does not save the changes, it loses this configuration sometimes and the box is unchecked. Prusa Slicer has the same behavior.

I take this opportunity to congratulate the developer, I started using the super slicer yesterday and I'm already a sponsor! The new ideas are really fantastic!

supermerill, 2020-08-31:
That also bothered me. I took a little time to investigate but i couldn't pinpoint the problem in the prusa code. I'll try again next time i touch on the preferences.

and thanks.

tokamac, 2020-11-06:
Maybe make the **Perspective Camera** view default instead of optional? If the setting constantly reverts for no reason it won't matter in this case. Isometric view is horrible to my eyes ;)

supermerill, 2020-11-06:
> Isometric view is horrible to my eyes ;)

and then someone else will complain that Perspective view is horrible for his eyes ;-p

tokamac, 2020-11-06:
Sure, but on a pure logical basis isometric view is made to easily align objects with a different depth location, it should be a temporary option rather than the opposite. Because when you manipulate a shape in three dimensions your brain actually expects this shape to have same proportions and behavior as a material object IRL. Imagine if VR and AR guys were making headsets made to work exclusively in isometric view without any perspective vanishing point. Holy Moses!

Rodrigormaraujo, 2020-11-06:
If it is possible to save the preferred viewing mode it will help a lot, so each person sets up how to like it more. This problem of not saving this configuration also happened with the prusa slicer. I didn't look at whether there was a correction because I started using the superslicer always.

supermerill, 2020-11-15:
If someone can reproduce the behavior in the 2.3.55, i'll fix it but right now i can't reproduce it anymore.

Rodrigormaraujo, 2020-11-15:
> If someone can reproduce the behavior in the 2.3.55, i'll fix it but right now i can't reproduce it anymore.

I'm already using it, very good, it was perfect. Thank you for this!

closed 2021-06-26 by supermerill