supermerill/SuperSlicer #435: Feature Request: Travel acceleration

SnowDrifterr, 2020-08-31:
I'd like to see an explicit travel acceleration implemented in acceleration control. I don't have that option for reprap firmware flavor within the slicer

supermerill, 2020-08-31:
?
machine->limits->travel acceleration ?

SnowDrifterr, 2020-08-31:
That doesn't seem to be reflected in my prints. For example, I have my outer perimeters set to a very low acceleration of 500mm/s^2. If the machine finishes an external perimeter then moves to a different spot, the resultant travel acceleration is very slow.

Looking closer at the gcode, it appears the generated code uses G1 for movements instead of G0. Perhaps that's part of it?

My original idea was to have something in the print profile for it under print settings -> speed -> acceleration control. For my machine limits, I keep them defined in the firmware so copying everything to the slicer seems redundant. But regardless, that still doesn't have the desired affect on travel moves.

supermerill, 2020-08-31:
Did you checked the "enable limits" that print the m205 in the starting section of the gcode?

SnowDrifterr, 2020-08-31:
Actually I think I found the issue. I did some prints sufficiently distanced apart on the bed to figure out what's going on.

My firmware uses mm/m for its nomenclature and the slicer lists as mm/s for speeds in the machine limits. The slicer outputs the value direct with no conversion. Always felt slow to me and now I know why. 

I'm in the RepRap flavor. Jerk, feed rates are defined in mm/min. Acceleration settings are mm^2/s. I think the slicer should be changed to reflect mm/m in the labels for homogeneity with the firmware.

supermerill, 2020-08-31:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M204_Set_printing_and_travel_accelerations

> Values are in mm/s².

Why do you need mm/min² ?

SnowDrifterr, 2020-09-01:
[image]

neophyl, 2020-09-01:
If you **know** that your machine firmware treats those values as mm/min then just scale the values and set them there.  Who cares what the labels say as long as the values passed to the machine are correct for that machine.
Or is there some reason that wouldn't work ?

supermerill, 2020-09-01:
Oh, ok, sorry, i didn't read that it's not about Travel acceleration anymore. Next time, you can open a new issue.

jerk limits:

https://duet3d.dozuki.com/Wiki/Gcode#Section_M205_Set_max_instantaneous_speed_change_in_mm_sec

feedrate:
I'll see to change the labels.

supermerill, 2021-08-09:
for travel accel setting:
commit 7ae60b6c82e3d6d4448044db6a3298b8f87189f4
should be in this night nigthly