supermerill/SuperSlicer #437: Negative estimated print time

7s3, 2020-08-31:
### Version
2.2.53

### Operating system type + version
Windows 10 Pro 64-bit. Version 2004

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When I slice STL file, the estimated print time is a large negative number.
Should return hours/minutes/seconds, but instead returns something like: -2147483648s

[image]

#### Project File (.3MF) where problem occurs
[fil_hng.zip](https://github.com/supermerill/SuperSlicer/files/5152401/fil_hng.zip)

ManuelMcLure, 2020-08-31:
I see this as well on occasion (and once yesterday a print said it was going to take over 9600 days). Usually moving the object on the bed, rearranging to the center and slicing again fixes the problem.

supermerill, 2020-08-31:
duplicate of #364

closed 2021-06-04 by supermerill