supermerill/SuperSlicer #439: Incorrect support speed , seems to stuck at some low speed.

kursatu, 2020-09-01:
### Version
_Version of SupserSlicer used goes here_
I've built with the latest commit on 8/27/2020
_Use `About->About SuperSlicer` for release versions_
2.2.53
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Custom CoreXY printer, Vking derivative
Marlin bugfix2, latest commit,
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Slice the object in the attached 3mf file.
Observe that the support speed it something completely different than anything in the settings.
I couldn't find where this speed comes from. It looks like a defect.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[image]
[image]
[Defect6.zip](https://github.com/supermerill/SuperSlicer/files/5153679/Defect6.zip)

neophyl, 2020-09-01:
Your speeds are being limited by the cooling over ride under filament settings.  If you reduce your layer time goal or the Max speed reduction it increases the support speed.  

Although I do think there is still a bug around that anyway, as I increased the min print speed to 50mm/s and the support was still printed at 20mm/s.  It took either reducing the layer time goal to be ridiculously short or changing the max speed reduction % to get it faster. It should only go as low as the min print speed but that didnt seem to work for the support when set to 50mm/s.

kursatu, 2020-09-01:
It seems like there is another related defect. If I switch supports (both interface and itself) to another extruder 
it does not slowdown for cooling. 
I usually print with just the support interface using extruder 4 but not both the support and interface.
For the original defect, the support is printed using the same tool as the model itself, so it should 
not slowdown to the cooling override speed. 
The rest of the layer does not  do that. 
This is a high impact defect. It makes the supports unusable.
[image]
[image]
[image]

supermerill, 2020-09-02:
Found the issue, it's because the support are on their own layer, and the algorithm consider that layer as a layer of the object, hence reducing the speed...

have to find a way to merge these support layer with the object's ones.

kursatu, 2020-09-02:
Thanks for looking into this. I am guessing it is not always fun to 
fix such defects, may be a regression.
I've tried to check if this happens with Prusa slicer but it did not even open the .3mf file.
I am guessing they can't be used between Prusa Slicer and SuperSicer.

neophyl, 2020-09-02:
Did you get a serialise error ? When trying to open in Prusa Slicer

supermerill, 2020-09-02:
I corrected some errors that can happen when loading config with unknown keys, but i can't do anything to put these corrections into prusaslicer.
Maybe i'll do an "export to prusaslicer" a day I'm bored.

supermerill, 2020-11-04:
the support-layer time will be added to the object-layer (if unsynch) in the 2.3.55 release.

kursatu, 2020-11-06:
I've got the latest code which includes the commit above, 2adea5b08d0a2f19352d414b98f2d5fbd9873993
The slowdown is still there.
This defect makes the support unusable.
[image]

kursatu, 2020-11-15:
Seems to be fixed with the latest sources on GItHub.

closed 2020-11-15 by kursatu