supermerill/SuperSlicer #442: min layer time broke

robthide37, 2020-09-01:
Newest update broke min layer time . I didn't bother loading a file cause i tried on two computers and multiple installs with pinhole edition

supermerill, 2020-09-02:
[image]

robthide37, 2020-09-02:
set min layer time to 60 sec and check speed reduction . It won't slow the print no matter if you have it set at 90 % and 10mms min speed it will just stay at the speed you have set

robthide37, 2020-09-02:
[image]
[image]

robthide37, 2020-09-02:
[image]

supermerill, 2020-09-02:
can i have your .3mf? (saveas -> zipped/renamed -> dropped here)

robthide37, 2020-09-02:
[Fan_Shraud_ELF (1).3mf.txt](https://github.com/supermerill/SuperSlicer/files/5165250/Fan_Shraud_ELF.1.3mf.txt)

robthide37, 2020-09-02:
diff one but still has the same prob

supermerill, 2020-09-03:
ok, it's an old switch (cooling) that i don't use anymore but was set in this filament profile to false.
The bug it that It's doing something, and it shouldn't.

For now, to re-activate it, you can add
```
setting:cooling
```
below
```
page:Cooling:time
group:Fan speed - default
```
in resources/filament.ui

closed 2020-10-22 by supermerill