supermerill/SuperSlicer #444: 2.2.53.1 Build Issue on Arch

pangklee, 2020-09-02:
### Version
2.2.53.1

### Operating system type + version
Arch Linux 5.8.5 / KDE 5.19.5

### Behavior
having issues building 2.2.53.1
```
[ 70%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/FreeCADDialog.cpp.o
In file included from /home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:9:
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/AppConfig.hpp:63:8: warning: extra tokens at end of #endif directive [-Wendif-labels]
   63 | #endif _NDEBUG
      |        ^~~~~~~
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘bool Slic3r::GUI::FreeCADDialog::load_text_from_file(const boost::filesystem::path&)’:
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:338:31: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
  338 |         catch (std::exception ex) {
      |                               ^~
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘bool Slic3r::GUI::FreeCADDialog::write_text_in_file(const wxString&, const boost::filesystem::path&)’:
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:379:27: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
  379 |     catch (std::exception ex) {
      |                           ^~
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘void Slic3r::GUI::FreeCADDialog::createSTC()’:
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:719:63: warning: conversion from ‘unsigned int’ to ‘wxColourBase::ChannelType’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Woverflow]
  719 |     m_text->StyleSetForeground(wxSTC_P_STRING, wxColour(128u, 256u, 0u)); // string, light green
      |                                                               ^~~~
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:721:66: warning: conversion from ‘unsigned int’ to ‘wxColourBase::ChannelType’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Woverflow]
  721 |     m_text->StyleSetForeground(wxSTC_P_CHARACTER, wxColour(128u, 256u, 0u));
      |                                                                  ^~~~
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp: In member function ‘void Slic3r::GUI::FreeCADDialog::test_update_script_file(std::string&)’:
/home/pkl/superslicer/src/SuperSlicer-2.2.53.1/src/slic3r/GUI/FreeCADDialog.cpp:802:27: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
  802 |     catch (std::exception ex) {
      |                           ^~
In file included from /usr/include/boost/asio/read.hpp:1288,
                 from /usr/include/boost/asio.hpp:112,
                 from /home/pkl/superslicer/src/SuperSlicer-2.2.53.1/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:64:
/usr/include/boost/asio/impl/read.hpp: In instantiation of ‘class boost::asio::detail::initiate_async_read_dynbuf_v1<boost::process::detail::posix::async_pipe>’:
/usr/include/boost/asio/impl/read.hpp:834:15:   required from ‘auto boost::asio::async_read(AsyncReadStream&, DynamicBuffer_v1&&, CompletionCondition, ReadHandler&&, typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type*) [with AsyncReadStream = boost::process::detail::posix::async_pipe; DynamicBuffer_v1 = boost::asio::basic_streambuf_ref<std::allocator<char> >; CompletionCondition = boost::asio::detail::transfer_all_t; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>; typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type = void]’
/usr/include/boost/asio/impl/read.hpp:808:20:   required from ‘auto boost::asio::async_read(AsyncReadStream&, DynamicBuffer_v1&&, ReadHandler&&, typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type*) [with AsyncReadStream = boost::process::detail::posix::async_pipe; DynamicBuffer_v1 = boost::asio::basic_streambuf_ref<std::allocator<char> >; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>; typename std::enable_if<(boost::asio::is_dynamic_buffer_v1<typename std::decay<_Tp2>::type>::value && (! boost::asio::is_dynamic_buffer_v2<typename std::decay<_Tp2>::type>::value))>::type = void]’
/usr/include/boost/asio/impl/read.hpp:850:20:   required from ‘auto boost::asio::async_read(AsyncReadStream&, boost::asio::basic_streambuf<Allocator>&, ReadHandler&&) [with AsyncReadStream = boost::process::detail::posix::async_pipe; Allocator = std::allocator<char>; ReadHandler = boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success<boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > > >::<lambda(const boost::system::error_code&, std::size_t)>]’
/usr/include/boost/process/detail/posix/async_out.hpp:136:32:   required from ‘void boost::process::detail::posix::async_out_future<p1, p2, Type>::on_success(Executor&) [with Executor = boost::process::detail::posix::executor<boost::fusion::joint_view<boost::fusion::tuple<boost::process::detail::posix::exe_cmd_init<char>, boost::process::detail::posix::io_context_ref>, boost::fusion::filter_view<const boost::fusion::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, boost::process::detail::posix::pipe_in&, boost::process::detail::posix::async_out_future<1, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::process::detail::posix::async_out_future<2, -1, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, boost::asio::io_context&>, boost::process::detail::is_initializer<mpl_::arg<-1> > > > >; int p1 = 1; int p2 = -1; Type = std::__cxx11::basic_string<cha [... cut ...]
make[1]: *** [CMakeFiles/Makefile2:1307: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
cmake from aur PKGBUILD file
```
                -DCMAKE_BUILD_TYPE=Release \
                -DCMAKE_INSTALL_PREFIX=/usr \
                -DOpenGL_GL_PREFERENCE=GLVND \
                -DSLIC3R_FHS=ON \
                -DSLIC3R_STATIC=OFF \
                -DSLIC3R_WX_STABLE=ON \
                -DSLIC3R_GTK=3 \
                -DSLIC3R_BUILD_TESTS=OFF

```

supermerill, 2020-09-02:
boost 1.70.0?

pangklee, 2020-09-02:
@supermerill boost 1.72.0-2

supermerill, 2020-09-02:
I'm using some more boost function now.The FreeCADDialog class use many new unused classes/functions from wxwidget and boost.

Maybe these have seen their name/package changed in 2 version?

That's why i build statically, to avoid these kind of problems.

pangklee, 2020-09-03:
Ok, Thanks! This is outside of the support for superslicer, but can you point me in the right direction to resolve my issue? I can use the appimage for now.

supermerill, 2020-09-03:
1) build the 1.70 version of boost: just build the deps and use them.
You can look at my script for building linux: https://github.com/supermerill/SuperSlicer/blob/master/.github/workflows/ccpp_ubuntu.yml

2) Or you can try to resolve the error, but i can't help you. Try to understand why boost don't like my code, resolve the error, and submit the patch (if applicable with boost 1.70).

khumarahn, 2020-09-04:
After a brief search, this is related to a bug in boost-1.72 which is fixed in boost-1.73, see https://github.com/boostorg/process/issues/116
I wonder why arch linux is still at boost-1.72 when the current version is 1.74. Of course 1.74 can have many new bugs :-) but arch is known for being up-to-date.

pangklee, 2020-09-05:
Yeah, I see 1.72 was flagged as out of date back in April. Aur has 1.75 but I suspect it’s going to have other issues. Thanks for looking into this. This is more of an arch issue so I’m closing the ticket.

closed 2020-09-05 by pangklee

Lukas1818, 2020-10-11:
I was able to compile SuperSlicer 2.2.53.3 with boost 1.71 on Arch.
Because the [boost packge](https://www.archlinux.org/packages/extra/x86_64/boost/) is outdated since over 5 month now, I decided to stop waiting for an official boost update and have updated the [aur package](https://aur.archlinux.org/packages/superslicer/) now.