supermerill/SuperSlicer #445: Seams not being placed on inside corners

VanessaE, 2020-09-03:
### Version

2.2.53 (git commit 06d2b9b2d0a5c823677aba1fb33d3e466aff351f)

### Operating system type + version

Debian bullseye

### 3D printer brand / version + firmware version (if known)

Custom-built Hypercube running Marlin, but that doesn't matter :stuck_out_tongue: 

### Behavior

Seams are not being placed on "inside" corners, rather "outside" corners and even overhangs are being chosen where inside corners would hide the seam better:

[image]

Pictured here are two views of the rear, right corner of a popular retraction test object (3mf below).  Here you can see that the outside corner is chosen for the seam position in this area of the object, until it gets up to where the overhanging part turns vertical, then the seam moves there.

The other three main corners demonstrate similar behavior.

The green line marks where I would have expected the seam to be placed along the entire height of the part.

"Nearest", "Aligned", "Rear", and "Corners" seam modes all produce this result.  Seam position objects don't help; even when I put one right next to an inside corner, a different spot is chosen anyway.  It's as if SS is expressly avoiding inside corners.  Shouldn't it try to avoid *outside* corners instead?

#### Project File (.3MF) where problem occurs

[Retraction.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5166310/Retraction.3mf.zip)

supermerill, 2020-09-03:
It's the wipe.

supermerill, 2020-11-01:
should be fixed on 2.3.55, as prusa reworked a bit the seam algo, I use that opportunity to tune it a bit more.

closed 2020-11-25 by supermerill