supermerill/SuperSlicer #446: Cutting Corners ++

DocRuzzy, 2020-09-03:
I just wanted to voice my support for the Cutting Corners Feature. I'm working to fabricate small single wall devices with switchback-overhangs and this looks like a possible solution to a sort of "lipping" I get on these parts. I'd love to see it further expanded to slow down print speed at corners as well. 

### Version
_Version of SupserSlicer used goes here_

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-09-03:
Thanks

> I'd love to see it further expanded to slow down print speed at corners as well.

It's a bit redundant with acceleration. Check your firmware, they should have a setting to slow down more at corners. See for example the [klipper way/explanation](https://github.com/KevinOConnor/klipper/blob/master/docs/Kinematics.md), and how to modify it in the config, with [square_corner_velocity (completely at the bottom)](https://github.com/KevinOConnor/klipper/blob/8ecec4be8ed3a0e88777b108d95f8a0b9d02dc7c/config/example.cfg)

kursatu, 2020-09-10:
+1 for @DocRuzzy  's proposal for slowing down at the corners.
My definition of the problem is that the small strips should not curl up. 
At such regions there should be enough time for cooling.

I don't think acceleration can and should really be used for cooling small strips.
It will increase the overall time quite a bit and wont adequately address the issue.
It requires another (probably complex) detection comparable to the overhang perimeters.

I am using cornering as well. Though, I am just not testing it rigorously yet, 
so not sure what the effect is. The promise is great though.
I am using other experimental features, like perimeter overlap.

nhopfen, 2020-09-29:
Having this at 100 seems to slightly improve square corners and seams compared to the best I could achieve previously with linear advance alone.

closed 2022-04-01 by supermerill