supermerill/SuperSlicer #450: G-Code flavor RepRap produces wrong M204 commands

toskium, 2020-09-05:
### Version
Superslicer 2.2.53

### Operating system type + version
Windows 10 pro

### 3D printer brand / version + firmware version (if known)
DIY Delta printer
Reprap Firmware 3.1.1

### Behavior
* gcode flavor is set to "RepRap"
* acceleration control is used to define different acceleration values
* model is sliced, gcode is produced
* acceleration values according to slicer are ignored

When looking up the relevant lines of gcode you will find:

`M204 Sxxxx`

According to the documentation over at RRF:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M204_Set_printing_and_travel_accelerations

The proper M204 command looks like:

`M204 Pxxx`

It seems to be a bug originally coming from the PrusaSlicer code since I can reproduce that behaviour on PrusaSlicer as well.

supermerill, 2020-09-05:
this is going to be changed for G10

supermerill, 2020-09-05:
I have 
```
M204 P500 T1500 ; sets acceleration (P, T), mm/sec^2
```

please post your project file for me to test.

toskium, 2020-09-05:
Here you go:
[testprint.zip](https://github.com/supermerill/SuperSlicer/files/5178506/testprint.zip)

closed 2020-10-22 by supermerill