supermerill/SuperSlicer #451: Issue with gcode output

atxbyea, 2020-09-05:
### Version
_Version of SupserSlicer used goes here_

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

 SuperSlicer 2.2.53 
### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

GalliumOS
Linux hinata 4.16.18-galliumos #1 SMP PREEMPT Sun Jun 23 04:14:45 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
buster/sid
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Ender3 Pro with latest Klipper
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

gcode tries to set extruder temperature with G10, see attached snippet here with my changes included : 
`M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S200
;G10 S200 ; set temperature
;G28 ; home all axes
;G1 Z5 F5000 ; lift nozzle
;G10 S200 ; set temperature
M116 ; wait for temperature to be reached
G28
G1 Z5 F5000
M109 S200
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
`
Also tries to do M116, which isn't valid for klipper either I think.

This only happens when using custom profile - klipper in superslicer
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

[file.zip](https://github.com/supermerill/SuperSlicer/files/5178553/file.zip)

supermerill, 2020-09-05:
you have gcode_flavor = reprap
not klipper

and reprap now use g10
https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Retract

qwewer0, 2020-09-15:
@atxbyea One solution is to set the G-code flavor to either `Klipper` or `Marlin`, and that will change `G10 S200` to `M104 S200`.
And don't forget to close this issue if your problem has been solved.

atxbyea, 2020-09-28:
Sorry for the late reply @qwewer0 @supermerill 

I was sure I used Klipper on my last try of superslicer, I have now downloaded the latest version of superslicer on a new machine and selected "klipper" as the custom printer, however I cannot see any indication of temperatures in the gcode, nor does the print start because it is complaining about extruder not being hot enough.

I want to move away from cura and superslicer seems to be the best choice out there.

See attached project and gcode.
[Atuin.zip](https://github.com/supermerill/SuperSlicer/files/5291438/Atuin.zip)

neophyl, 2020-09-28:
I'm probably not the best at tracing these things but the thing I noticed about your Printer setup is you dont have any commands to set the extruder temp or bed temp initially in your start gcode.  As your First layer and Other layer temp settings under filament are also the same at 210 degrees it doesnt insert a temp command.  As an experiment I changed the other layers temp to 220 and it inserted a M104 S220 at what looks like the second layer to me. 
Normally in the start gcode you would use something like -
M140 S[first_layer_bed_temperature] ; set bed final temp
M104 S[first_layer_temperature] ; set extruder final temp
M109 S[first_layer_temperature] ; wait for extruder final temp
M190 S[first_layer_bed_temperature] ; wait for bed final temp
(for Marlin, substitue the corretc code for Klipper).  

These set your initial extruder and bed temps from the values set under Filament - first layer.  The the slicer inserts any changes to those IF DIFFERENT on subsequent layers.  As your values are the same then the gcode doesnt contain any commands to set either the extruder or bed temp.

I think you need to add the lines I posted above to your printer start gcode.  Or use a Klipper macro to set the temps triggered by something added to your start gcode.  Personally I'd just use the simple way.

supermerill, 2020-09-28:
You have to write the m104 yourself for klipper, as most of the klipper user use their own macro instead, and so i can't detect that you forgot to write it.

atxbyea, 2020-09-28:
> You have to write the m104 yourself for klipper, as most of the klipper user use their own macro instead, and so i can't detect that you forgot to write it.

Interesting, good to know, I was expecting the slicer to fill them when the fields are populated under settings, but I can try adding them myself and see if it starts printing as expected.

atxbyea, 2020-09-28:
> I'm probably not the best at tracing these things but the thing I noticed about your Printer setup is you dont have any commands to set the extruder temp or bed temp initially in your start gcode. [...]

Thanks for the examples, I'll go ahead and try to add it at the start of my gcode and see what happens.

atxbyea, 2020-09-30:
Issue solved, works fine by adding from example 👍

closed 2020-11-16 by atxbyea