supermerill/SuperSlicer #452: SuperSlicer crashes when clicking on Gcode Thumbnails settings.

LazaroFilm, 2020-09-06:
### Version
2.2.53

### Operating system type + version
MacOS 10.15.6

### 3D printer brand / version + firmware version (if known)
HEVO / Klipper

### Behavior
SuperSlicer crashes when clicking on Gcode Thumbnails: setting. The whole window closes instantly. 

#### Project File (.3MF) where problem occurs
Happens without any opened project.

supermerill, 2020-09-06:
maybe something related to #222

Can't replicate

LazaroFilm, 2020-09-08:
Yes, the value in the config file was `thumbnails =`
I had imported settings from PrusaSlicer for this printer, that might be the reason why my values were empty? Maybe implementing a check on this value, if it's empty, make it equal to the default value?

supermerill, 2020-09-08:
The check is done, in the nightly build.

closed 2020-09-20 by LazaroFilm