supermerill/SuperSlicer #453: "Bridge-" and "Above bridge flow ratio" - Cannot get desired results

pandaym3D, 2020-09-07:
### Version
2.2.53

### Operating system type + version
Windows 10 18362

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with Klipper

### Behavior
When I go through the calibration processes, I cannot seem to get the bridge and above bridge flow ratios tuned.

These pictures show the bridge float ratio prints:
[image]
[image]

Hardly any difference on them. So I keep it at a 100%, and move onto the above bridge. Here I couldn't see any difference, so I printed a couple of extra at 135, 140, 150 and 165 too. They don't look the same, but none of them are better than others or differe much:
[image]

I don't know if the functionality doesn't work for me, or if I am doing something wrong?

supermerill, 2020-09-07:
125 is much better than the 100.
It's not perfect yet. there is a lack of filament or the bridge are really too "droopy".
- flow calibration has been done?
- pla, abs?
- you can also check infill->reducing printing time -> supporting dense layer with automatic algo. This one will help to support the bridge before top layers.

neophyl, 2020-09-07:
I did some tests too, I found with using support below bridges and 120-130% on the bridge ratio I could get really nice bottom layers, but the support was sometime a bit harder to remove.

The 'worst' aspect I found (but still not too bad) was that the perimeters on the bridge areas (dark blue on the previews) were still spaced apart slightly so for a test overhang area with support below the perimeter would sometime detach from the rest by a strand.  Still way better than the standard though.

pandaym3D, 2020-09-07:
> 125 is much better than the 100.
> It's not perfect yet. there is a lack of filament or the bridge are really too "droopy".
> 
> * flow calibration has been done? To be extra sure, extrude a 30mm cube with 50% rectilinear infill and weight it on a precision scale vs the weight given by the slicer.
> * pla, abs?
> * you can also check infill->reducing printing time -> supporting dense layer with automatic algo. This one will help to support the bridge before top layers.

Flow calibration is done yes. I will post all my calibration results below here. I don't have a precision scale at hand, so cannot really do a weighing test. I have my E-steps calibrated for my Bondtech extruder, and before I started using Klipper and SuperSlicer, I found that I almost always had to use 1.0 EM as per regular EM testing (without the Prusa standard 0.95 EM start GCODE). But I find now that I need it slightly higher. Probably no biggie though.

I actually enabled supporting dense layer (auto) with the 145 and 165 over bridge flow samples I did, it didn't seem to make a difference in this test.

I agree that 125 is more filled than 100, but it doesn't appear to be linear how filled they are depending on the above bridge flow rate. Also the 125% and 165% look almost identical. It is very likely that it is just me that does not understand exactly how these things work :)

The material is 3D870 PLA - 3D Fuel Pro: https://eu.3dfuel.com/collections/pro-pla/products/apla-plus-fossil-fuel-black.

I have done all the other calibrations:

For flow calibration, I started at 1.0 and ran the coarse test with these results:
[image]
[image]
[image]
[image]
[image]

I see that -10 is slightly underextruded and +10 is over extruded. So I set the EM to 1.08 and run the fine test with these results:
[image]
[image]
[image]
[image]
[image]

The best one to me looks like it is slightly below the -4. So 1.08*0.96 = 1.037 and I therefore set it to 1.03.

For the temperature calibration I ran 200-230 in steps of 5:
[image]
[image]

The lines you see are not visible in real life, but I am shining a 3200 lm flashlight at it to get it to pick up the details. But I think 210 looks like a good temperature, although the material does seem very tolerant to temperatures.

So far I am pretty confident tuning those things, but when we get to the bridging, I start to be confused. I set both flow ratios for bridging to 100% and ran the first bridge flow test:
[image]
[image]
[image]
[image]

This one I think is extremely hard to judge, but I decided 100% looked like the cleanest bottom, even though they were very similar. I then ran the over bridge flow tests with the results above.

supermerill, 2020-09-07:
mmm
i don't think it's wise to go over 125% on over-bridge.
You can try the over-bridge calibration with a bigger bridge ratio like  neophyl  said. With a supporting layer from the "dense infill", it may push the bridge layer a bit higher up.
If it doesn't work, I'm afraid that the only way for you is to add a bit of over-extrusion on the top layer with the top layer flow. 
Maybe I'll have to rework a bit the over-bridge feature to span more than a single layer...

neophyl, 2020-09-07:
I wasn't exactly testing the same thing and I wasn't using any inbuilt tests, I was trying when the bridge % was first added.  My test piece was a short pillar with a 1" long, roughly 1/2" wide overhang section.  So not actually a bridge.  With support under the section I wanted to see what sort of bottom surface I could get as traditionally Slic3r/Prusa Slicer produces pretty bad bottoms in that situation.  
With support for the bridge layers to press onto and around 120% flow on the bridges it actually gave a very nice surface.   You just have to make sure that the support is at 90 degrees to the direction of bridging or it is impossible to remove the support and the surface looks worse.  Within those bounds though you can greatly improve the bottom layer.  I did also try with a bridging test part from thingiverse and if you force supports for those the same bottom surface can be achieved.  So not a bridge as such but the flow adjust is certainly useful for those scenarios.