supermerill/SuperSlicer #454: strange slicer behavior

yanfali, 2020-09-07:
### Version
_Version of SupserSlicer used goes here_
2.2.53
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
MacOS 10.14.6

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Custom bedslinger running klipper

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
Slices the inverse of what I'm expecting for model

Expected (via cura 4.6)

[image]

Actual (superslicer):

[image]

Model Mode (super slicer):

[image]

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

[Buchstütze_-_SeiteMitVerbinder_w.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5184925/Buchstutze_-_SeiteMitVerbinder_w.3mf.zip)

supermerill, 2020-09-07:
[image]
cannot reproduce.
99% of chance that's because the stl is bad (see the warning near it). please repair it (try the repair tool in filemenu)

yanfali, 2020-09-08:
Hi Merril, not getting a repair option. Attaching the STL for your perusal. Not entirely sure what I'm doing wrong here, but yes, probably just a bad STL.

[Buchstütze_-_SeiteMitVerbinder_w.stl.zip](https://github.com/supermerill/SuperSlicer/files/5189500/Buchstutze_-_SeiteMitVerbinder_w.stl.zip)

supermerill, 2020-09-08:
i can reproduce with your stl.
If i use the repair tool (it uses a microsoft util, so maybe only available on windows), i get this obj, and it sliced correctly.
[image]

yanfali, 2020-09-08:
Thanks. Closing.

closed 2020-09-08 by yanfali