supermerill/SuperSlicer #456: Wipe into infill causes MCU overload on Klipper

cab404, 2020-09-08:
### Version
2.2.53.1

### Operating system type + version
NixOS

### 3D printer brand / version + firmware version (if known)
TronXY X5SA with Klipper

### Behavior
I enabled 'Wipe while retracting' and 'Firmware retraction' (reproduces without the latter). SS generated following retraction code:
```

G1 X161.946 Y190.560 E0.06376
G92 E0
G10 ; retract
G1 F6240.000
M204 S1500
G1 X161.946 Y192.083 E-1.34781
G1 X144.806 Y187.782 F7800.000
G11 ; unretract
G1 F3000.000

```

As you can see, it generates E-move for ~1.3 mm at about 104 mm/s, which equates to 
1.34781 * (6240 / 60 / 0.001181952) = 118593 steps per second on extruder alone.

#### Project File (.3MF) where problem occurs

[Base_Mount.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5191204/Base_Mount.3mf.zip)

cab404, 2020-09-08:
Can confirm that disabling 'Wipe into infill' definitely fixed mcu overload.

1N4148, 2020-09-10:
I don't see any E-only move with 104mm/s as this is the code with firmware retraction enabled. Can you post the code without firmware retraction but with wipe while retracting instead?

cab404, 2020-09-11:
> I don't see any E-only move with 104mm/s as this is the code with firmware retraction enabled. Can you post the code without firmware retraction but with wipe while retracting instead?

Oof, sorry for not supplying more data. Let's go! :D

I've written a script to crudely analyze steps per second on extruder, and ran it on two gcode files: one with wipe enabled, and one with it's left disabled. I found several places, where values jump into tens of millions of steps per second on cinematic moves.

Here's a script I made, that outputs all e-moves with step count > 10000:

```python3
#!/usr/bin/env

# GCode steps per second analyzer and grapher

import os, sys, typing as t, math

fname = sys.argv[1]
print(fname)

gcode = open(fname, "r").readlines()

def read_gcode_line(line: str):
    code = line.split(";", 1)[0].strip()
    if not len(code):
        return None
    
    command = code.split(" ", 1)
    if len(command) == 1:
        return command[0], None
    else:
        return command[0], command[1].split(" ")

def read_move_attrs(attrs: t.List[str]):
    defaults = {
        "X": None,
        "Y": None,
        "Z": None,
        "E": None,
        "F": None
    }
    defaults.update(dict(map(lambda x: (x[0], float(x[1:])), attrs)))
    return defaults

x_mmps = 0.006275
y_mmps = 0.006275
z_mmps = 0.00125
e_mmps = 0.001172184

x = 0
y = 0
z = 0
e = 0
f = 5000
absolute = True

MOVE = ["G0", "G1"]
SET = ["G92"]
ABS = ["G90"]
REL = ["G91"]

for number, line in enumerate(gcode):
    cmd = read_gcode_line(line)
    if cmd is None: continue
    cmd, attrs = cmd
    dx, dy, dz, de = x, y, z, e

    if cmd in ABS:
        absolute = True
    elif cmd in REL:
        absolute = False
    elif cmd in SET:
        # don't calculate delta here
        move_attrs = read_move_attrs(attrs)
        x = move_attrs["X"] or x
        y = move_attrs["Y"] or y
        z = move_attrs["Z"] or z
        e = move_attrs["E"] or e
        f = move_attrs["F"] or f
    elif cmd in MOVE:
        move_attrs = read_move_attrs(attrs)
        if absolute:
            x = move_attrs["X"] or x
            y = move_attrs["Y"] or y
            z = move_attrs["Z"] or z
            e = move_attrs["E"] or e
        else:
            x += move_attrs["X"] or 0
            y += move_attrs["Y"] or 0
            z += move_attrs["Z"] or 0
            e += move_attrs["E"] or 0
        f = move_attrs["F"] or f
    dx -= x
    dy -= y 
    dz -= z 
    de -= e
    mmps = f / 60
    dst_cinematics = math.sqrt((dx ** 2) + (dy ** 2) + (dz ** 2))
    time_max = dst_cinematics / mmps

    if (int(abs(de) / e_mmps)) > 10000:
    
        print(f"{sys.argv[1]}:{number+1}: speed {mmps}")
        
        if dst_cinematics != 0:
            time = dst_cinematics / mmps
            print(f"cinematics move: {dst_cinematics / mmps:.3f} sec")
        else:
            time = abs(de) / mmps
            print(f"e only move: {abs(de) / mmps:.3f} sec")
        
        print(f"steps   {int(abs(de) / e_mmps)} ")
        print(f"steps/s {int(abs(de) / e_mmps) / time:.3f} ")
        print()
```

Here's all the files I used in this test:
[overload.zip](https://github.com/supermerill/SuperSlicer/files/5206797/overload.zip)

Here's output for wipe.gcode:
```
wipe.gcode
wipe.gcode:32: speed 5.0
e_only_move: 3.000 sec
steps   12796 
steps/s 4265.333 

wipe.gcode:51: speed 25.0
e_only_move: 0.720 sec
steps   15355 
steps/s 21326.389 

wipe.gcode:456: speed 64.0
time_accel_inf 0.002 sec
steps   39956 
steps/s 16843639.373 

wipe.gcode:2431: speed 64.0
time_accel_inf 0.001 sec
steps   121792 
steps/s 98281870.535 

wipe.gcode:7918: speed 64.0
time_accel_inf 0.003 sec
steps   20932 
steps/s 7216921.287 

wipe.gcode:59077: speed 64.0
time_accel_inf 0.016 sec
steps   817326 
steps/s 50402925.320 

wipe.gcode:64356: speed 64.0
time_accel_inf 0.005 sec
steps   68818 
steps/s 12636478.253 
```

Here's for nowipe.gcode:
```
nowipe.gcode
nowipe.gcode:32: speed 5.0
e only move: 3.000 sec
steps   12796 
steps/s 4265.333 

nowipe.gcode:51: speed 25.0
e only move: 0.720 sec
steps   15355 
steps/s 21326.389 

nowipe.gcode:278: speed 25.0
e only move: 0.504 sec
steps   10759 
steps/s 21326.793 

nowipe.gcode:440: speed 25.0
e only move: 1.991 sec
steps   42468 
steps/s 21327.620 

nowipe.gcode:2410: speed 25.0
e only move: 5.829 sec
steps   124326 
steps/s 21327.645 

nowipe.gcode:7802: speed 25.0
e only move: 1.099 sec
steps   23432 
steps/s 21327.069 

nowipe.gcode:58959: speed 25.0
e only move: 38.427 sec
steps   819556 
steps/s 21327.689 

nowipe.gcode:64213: speed 25.0
e only move: 3.341 sec
steps   71248 
steps/s 21327.583 
```

cab404, 2020-09-11:
It of course relies on that E-moves are fitted into cinematic moves.

1N4148, 2020-09-11:
You are using 32 microsteps & pressure advance with a bowden setup?

cab404, 2020-09-11:
> You are using 32 microsteps & pressure advance with a bowden setup?

I left microstep settings at default, so that might be the case?

1N4148, 2020-09-11:
So, it's more likely that pressure advance is overloading your extruder stepper. What are your pressure advance settings? Also, TMC2208 on extruders still have problems, especially if stealthchop is used.

cab404, 2020-09-11:
> So, it's more likely that pressure advance is overloading your extruder stepper. What are your pressure advance settings? Also, TMC2208 on extruders still have problems, especially if stealthchop is used.

I went through trouble of reprogramming my e driver to disable stealthchop. Pressure advance is at 1.5.

1N4148, 2020-09-11:
Pressure advance 1.5 is way to high, even for a bowden setup. And you're not the first one who's running into trouble using a TMC2208 with pressure advance (Klipper) or linear advance (Marlin).

You cannot limit extruder speeds & acceleration on pressure advance moves, there's no limit. And wipe while retracting is making a lot of retraction moves which are causing pressure advance to make a lot of nonlinear extruder moves causing the TMC2208 to fail.

There are some options to test:
* Disable wipe while retracting & not using firmware retraction (there's no advance in using firmware retraction over the slicers one) 
* Lower pressure advance < 1.0, trying to increase "smooth_time" to 0.1 or 0.2
* Try "retract amount before wipe" with about 80%

cab404, 2020-09-11:
> Pressure advance 1.5 is way to high, even for a bowden setup. And you're not the first one who's running into trouble using a TMC2208 with pressure advance (Klipper) or linear advance (Marlin).

These troubles usually resulted in stepper turning off, not controller overloading. IMO this is unrelated :shrug: 

> You cannot limit extruder speeds & acceleration on pressure advance moves, there's no limit. And wipe while retracting is making a lot of retraction moves which are causing pressure advance to make a lot of nonlinear extruder moves causing the TMC2208 to fail.

That might be the case if speeds were changing dramatically, but that's really not — it's infilling at 150 with 4k accel just fine, and then stepper gets far into the past on some gap infill at 50 :|

> Lower pressure advance < 1.0, trying to increase "smooth_time" to 0.1 or 0.2

I will try retuning LA with those, thank you for the suggestion!

> Disable wipe while retracting & not using firmware retraction (there's no advance in using firmware retraction over the slicers one)

I used it mostly for easier navigation through gcode while debugging.

> Try "retract amount before wipe" with about 80%
I will try that if LA suggestions won't help. Thank you! :з

cab404, 2020-09-11:
Closing the issue as it probably not related to slicer itself.

closed 2020-09-11 by cab404

1N4148, 2020-09-11:
"and then stepper gets far into the past on some gap infill at 50 :|"

Check "smooth_time". Well known issue in klipper

supermerill, 2020-09-11:
Is there a problem on the retraction speed with "Firmware retraction" vs not "Firmware retraction" ?

1N4148, 2020-09-11:
> Is there a problem on the retraction speed with "Firmware retraction" vs not "Firmware retraction" ?

Nope, I meant that with retraction settings in slicer there's more flexibility than using firmware retraction

n8bot, 2020-09-11:
I think this very well may be an issue with the way Slic3r (or maybe just PrusaSlicer) variants calculate the E amount for a segment.

The E amount is calculated with a length-value that is calculated from X/Y positions that have much higher precision than those outputted in GCode. The result is that the E amount to extrude is not exactly correct for some moves, especially moves that are very small. It's possible that the very small "wipe-while-retract" moves are being generated with a sensible calculation behind the scenes, but the outputted GCode has been rounded and invalidates the "somewhat reasonable" extrusion amount associated with the wipe-while-retract move. I.e., since the firmware is trying to synchronize the wipe-while-retract move, because the extrusion takes place alongside the X/Y movement, the rounded X/Y coordinates might make that move exceptionally ridiculous to perform. Add pressure advance to that mix, and you're in for a fun time.

cab404, 2020-09-24:
> "and then stepper gets far into the past on some gap infill at 50 :|"
> 
> Check "smooth_time". Well known issue in klipper

Smooth time helped reducing step count — thank you — though degraded LA on multiple consecutive corners a lot.
I settled on adding a direct extruder on my printer)