supermerill/SuperSlicer #457: Problems starting superslicer

MalsakerLures, 2020-09-09:
Hi all,

I just ran the compilation of superslicer following the instructions shared here and had the pre-requisites installed.

after compiling i found this outcome in my terminal window on MacOS:

20 warnings generated.
[ 91%] Linking CXX static library liblibslic3r_gui.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: liblibslic3r_gui.a(FixModelByWin10.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: liblibslic3r_gui.a(FixModelByWin10.cpp.o) has no symbols
[ 91%] Built target libslic3r_gui
make: *** [all] Error 2

Did it not finish, or if it did where do i find the software?

Thank you!

supermerill, 2020-09-09:
There were an error when compiling, search for 'error' to found the message. Cannot help you without it.

Also, you should rename the issue with "compiling on maccos" instead of "starting" . If you just want to start, there is the releases and the nightly builds.

MalsakerLures, 2020-09-09:
Thank you, however i had to reboot and lost my Terminal-window.
How do i remove what ever i have done previously and start again fresh?

Do simply remove the folder structure and start from scratch?

supermerill, 2020-09-09:
you don't need to start "fresh".
just run the last command again.

If you really want to start "fresh", delete the build directory and start again from the cmake.

MalsakerLures, 2020-09-09:
Ahh, ok, well i will start from scratch..:)

But, is there a difference in the ui intefaces if i use xcode or not?

Cheers!

Skickat från min iPhone

MalsakerLures, 2020-09-09:
So im running a new install and when i compile the cmake dependencies i get these warning initially:
[  5%] Performing build step for 'dep_wxwidgets'
./src/osx/core/bitmap.cpp:1362:29: warning: unused parameter 'w' [-Wunused-parameter]
void wxBitmap::SetWidth(int w)
                            ^
./src/osx/core/bitmap.cpp:1368:30: warning: unused parameter 'h' [-Wunused-parameter]
void wxBitmap::SetHeight(int h)
                             ^
./src/osx/core/bitmap.cpp:1374:29: warning: unused parameter 'd' [-Wunused-parameter]
void wxBitmap::SetDepth(int d)
                            ^
3 warnings generated.
./src/common/dcbufcmn.cpp:88:5: warning: '__WXMSW__' is not defined, evaluates to 0 [-Wundef]
#if __WXMSW__
    ^
1 warning generated.

MalsakerLures, 2020-09-09:
Anyhow, will continue thru and see if any other issues occur and then i will post them here.

MalsakerLures, 2020-09-09:
@supermerill ran thru it again and got the same error. copied everythin to this pastebin:
https://pastebin.pl/view/e20e9a2c

supermerill, 2020-09-09:
same problem as https://github.com/supermerill/SuperSlicer/issues/432

you have to set the cmake variable SLIC3R_BUILD_TESTS to OFF

MalsakerLures, 2020-09-09:
not sure how to change that thou, where do i change that parameter?

supermerill, 2020-09-09:
if you open cmake-gui, and put the path to your build folder,it should show you all the parameters, and you can modify them, save, click generate and then do your make.

MalsakerLures, 2020-09-09:
So sorry, but i just realised, adn what you mentioned earlier, the pre-compiled package......that is what im going for since im no developer. But i really appreciate you support.

However, if it would help then i can try it again if you need me to for problemsolving purposes?

supermerill, 2020-09-09:
thank you. no need for now.

closed 2021-06-04 by supermerill