supermerill/SuperSlicer #459: Bug - Disable only for outter side

VectorPort, 2020-09-09:
### Version
2.2.53

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
-

### Behavior
If you enable External perimeter first, then only for outter side and only for inner side are enabled. Then, if you turn off only the only for outter side, the program is immediately closed. All this happens if there is a model on the table. If there is no model, nothing happens.

#### Project File (.3MF) where problem occurs
-

supermerill, 2020-09-09:
If you save your project, close the slicer, start again, open your project then uncheck the fireld, does it crash?
If yes, can you post your project file (zipped)?

VectorPort, 2020-09-09:
Recorded a video when an error occurs. Maybe it will be easier to understand. The error occurs with any file.
[bandicam 2020-09-09 15-51-32-196.zip](https://github.com/supermerill/SuperSlicer/files/5195260/bandicam.2020-09-09.15-51-32-196.zip)

supermerill, 2020-09-09:
yet, it doesn't crash on my computer with my settings.

neophyl, 2020-09-09:
Doesn't crash on mine either.  Same version , windows 10.

VectorPort, 2020-09-09:
Deleted the entire folder with settings, and restarted. No more crashes. Sorry for the fake call)

VectorPort, 2020-09-09:
I tried it on another PC. As soon as I press Slicer now, he throws an error.

closed 2021-06-26 by supermerill