supermerill/SuperSlicer #461: Incorrect retract length, then correct unretract , results in massive blob.

kursatu, 2020-09-10:
### Version
_Version of SupserSlicer used goes here_
The latest, built against the commit from  Sep 7, 2020
_Use `About->About SuperSlicer` for release versions_
2.2.53
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
### Behavior
* _Describe the problem_
- Slice the attached project.
- Observe that the wipe tower retract is incorrect. It uses extruder retract length
, rather than filament override retract length.
- Subsequent unretract is correct, which results in massive blob.

This is what is generated.
; CP TOOLCHANGE END
;------------------
; printing object Large Hoop skinny_4mm_inner.stl id:0 copy 0
G1 E-0.60000 F1200.00000 
G1 F16800.000
G1 X151.839 Y297.681 E-0.38000
G1 E-0.02000 F1200.00000
G1 X253.918 Y278.453 F21000.000
G1 E4.00000 F1200.00000

Released version generates correct retract length
G1 E-3.60000 F1200.00000 

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[Defect8.zip](https://github.com/supermerill/SuperSlicer/files/5198634/Defect8.zip)

supermerill, 2020-09-10:
it's not exactly that it uses the wrong number, it's that there is a new retraction before changing the extruder, just after the header.
I'm looking at why this regression has happened...

robthide37, 2020-09-11:
try changing from relative ? low microsteps can cause this afaik

kursatu, 2020-09-11:
There is a nullptr crash. I've hacked my way around it for now. Please see callstack in the comments of the commit.
It could reproduce for any single material print.

kursatu, 2020-09-11:
The reported defect seems to be fixed though. 
Thank you for the fast turn around.. 
You seems to be really committed to make this slicer better.

kursatu, 2020-09-12:
The latest commit seems to have fixed the issue. The crash also did not happen.
Thank you.

closed 2020-09-12 by kursatu