supermerill/SuperSlicer #462: Unwanted holes in sliced object with 100% infill

nilrog, 2020-09-10:
### Version
v2.2.53

### Operating system type + version
macOS Mojave 10.14.6

### 3D printer brand / version + firmware version (if known)
Custom printer with Duet3D controller running RRF v3.3.1.
Same happens if I slice the model for Prusa Mini.

### Behavior
* _Describe the problem_
Slicing the STL with 3 perimeters and 100% infill leaves holes in the generated output between the screw holes in the corners and the external shell. This results in a weakened 3D-printed object.
Same model sliced in Cura with 3 perimeters and 100% infill is solid where the gaps are when sliced with SuperSlicer.
* _Steps needed to reproduce the problem_
See linked .3MF file below.
* _Expected Results_
Output filled to 100%

* _Actual Results_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Infill issue.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5201771/Infill.issue.3mf.zip)

supermerill, 2020-09-10:
seems like duplicate of #448

nilrog, 2020-09-10:
Yes, that is the same issue. I was focusing on infill so I didn't find the existing issue.
The gap is closed if I increase layer height to 0.25.

Feel free to close this one if you want. I can track the other issue instead.

closed 2020-09-10 by supermerill