supermerill/SuperSlicer #465: Printing with nozzle settings smaller than 0.4mm introduces odd warnings related to layer height/extrusion width in 'Print Settings'

ayerlock, 2020-09-12:
### Version
2.2.53

### Operating system type + version
Windows 10 Pro v2004 build:19041.450

### 3D printer brand / version + firmware version (if known)
Voron v2.4/Klipper

### Behavior
* Under Print Settings -> Perimeters & Shell -> Vertical Shells I am seeing an odd warning when using a 0.25mm nozzle and a 0.125mm layer height.

`Recommended object min (thick) wall thickness for layer height 0.125 and 1 perimeter: 0.5mm, 2 perimeter: 1.02mm, 3 perimeter: 1.52mm, 4 perimeter: 2.03mm, 5 perimeter: 2.54mm`

This would seem to indicate that I need to set a minimum of a 200% Perimeter/External Perimeter extrusion width under Print Settings -> Width & Flow -> Extrusion Width.  First, that seems insanely high.  Second, even if I make those changes to the extrusion widths it does not fix the warning.

### To reproduce the issue:
- Set nozzle size under Printer Settings to 0.25mm
- Set 'Base Layer Height' to 0.125mm under Print Settings -> Slicing
- Check 'Perimeters' under Print Settings -> Perimeters & Shell -> Vertical shells

* Normal Slicing
* Warning under Print Settings -> Perimeters & Shell -> Vertical shells
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs

[voron_design_cube_v6.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5212939/voron_design_cube_v6.3mf.zip)

supermerill, 2020-09-13:
It just indicate the needed thickness for an object to be printed with the said number of loop of perimeters.

ayerlock, 2020-09-16:
It doesn't indicate what settings you need to check to verify that you are getting what you want.  

More explicitly it should say something along the lines of "Thin walls will be used unless 'Print Settings' -> 'Width & Flow' -> 'Extrusion Width' -> 'Perimeters' >= 0.5mm (or whatever setting that is referring to [i.e. Default Extrusion width, etc.].