supermerill/SuperSlicer #466: Feature Request: Wipe after retract (for firmware retraction)

1N4148, 2020-09-12:
Hi,

as "firmware retraction" and "wipe while retract" are mutually exclusive, do you think it would be possible to add an "wipe after retract" for those who'd like to use firmware retraction? 

For me, the big benefit using firmware retraction is having the possibility to live tune the parameters.

paukstelis, 2020-09-26:
Slightly off-topic for this FR, but would requires changes in the same area: how about moving the `; stop printing OBJECT` comments to the end of any wipe moves? Object cancelling that uses these stop comment tags would allow all wipes to occur, which is a bit of waste.

supermerill, 2020-09-28:
It's a bit difficult as the wipe may be done at the start of an other extrusion. Before this ither extrusion, the writer don't know if it will need a mive or a z change.
It's possible to save the comment for later, but much more complicated to ensure it will be printed.

1N4148, 2020-09-28:
Ok, I thought it would be easy to trigger a firmware retraction and then do the wipe moves just with no E-Value.