supermerill/SuperSlicer #467: Strange layer chang retraction behaviour

1N4148, 2020-09-13:
### Version
Actual master

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro / Klipper (master)

### Behavior
With "Retract on layer change = False" the end of layer retraction is done after the z-movement. 

`G1 X115.787 Y99.618 E0.01022
G1 X115.399 Y99.751 E0.01095
G1 X115.147 Y99.537 F12000.000
; stop printing object Stringing_Test.stl id:0 copy 0
;BEFORE_LAYER_CHANGE
;12.2

G1 Z12.200000 F12000.000
;AFTER_LAYER_CHANGE
;12.2
; printing object Stringing_Test.stl id:0 copy 0
G10 ; retract
G1 X114.942 Y121.281
G11 ; unretract
G1 F1500
G1 X115.096 Y121.285 E0.00410
G1 X115.245 Y121.321 E0.00410
G1 X115.383 Y121.388 E0.00410`

With "Retract on layer change = True" the end of layer retraction is done before the z-movement.

`G1 X116.111 Y99.415 E0.01022
G1 X115.787 Y99.618 E0.01022
G1 X115.399 Y99.751 E0.01095
G1 X115.147 Y99.537 F12000.000
; stop printing object Stringing_Test.stl id:0 copy 0
;BEFORE_LAYER_CHANGE
;12.2

G10 ; retract
G1 Z12.200000 F12000.000
;AFTER_LAYER_CHANGE
;12.2
; printing object Stringing_Test.stl id:0 copy 0
G1 X114.942 Y121.281
G11 ; unretract
G1 F1500
G1 X115.096 Y121.285 E0.00410
G1 X115.245 Y121.321 E0.00410
G1 X115.383 Y121.388 E0.00410`

I've changed to firmware retraction for better visualization of the retract/deretract. The retraction is correct, cause it's a layer change with travel to the new start position. But on raising z the first variant (retract after z-movement) leaves a blob. 

Without firmware retraction but using "Wipe while retracting" it's getting more strange: Without "retract on layer change" there's no wipe done before leaving the layer and travelling to the new start. Wipe will only happen within the same layer. 

Don't know if this behaviour is wanted but it took me hours to figure out where the stringing is coming from.

supermerill, 2020-10-29:
There is a retraction before a travel.
If the retract_layer_change is set, a retraction is enforce before the z-move, and so the retraction before the next travel won't do anything (more).

What are your proposal to makes it more obvious? some tooltip modifications to propose?

1N4148, 2020-10-29:
For me the logic is a bit confusing as I'd expect any retraction would be always done before lifting Z. But from the logic's side it's clear as the Z-movement is smaller than the travel retraction threshold.

closed 2020-10-31 by 1N4148